arendst/Tasmota #2026: Sonoff dual gpio14

andyd999, 2018-02-25:
If I connect a temp sensor to gpio14 on a sonoff dual will that affect both channels operation?
I wish to use on my aquarium so channel 1 controls the lighting and channel 2 controls the heater based on the input from the temp sensor. However I need the sensor to work on only 1 channel

curzon01, 2018-02-26:
as far as I know in Tasmota there is no relation between a sensor and a power channel at all.
To control a power channel depending on a certain e. g. temperatur value from a sensor you need an external system which creates this relation. So your question superfluous

andyd999, 2018-02-26:
Well on say a th10 tasmota will control the relay based on a rule with input from the temperature probe. I just wondered on a 2 channel If I connected to gpio14 which electrical relay would be influenced

curzon01, 2018-02-26:
>  th10 tasmota will control the relay based on a rule with input from the temperature probe
Can you pls show me the rule? Where you read it?

andyd999, 2018-02-26:
Actually the one I have working uses stock firmware not tasmota my bad. 
I'll flash it and see if home assistant scenes can create the same effect 
Failing that I'll just use one th10 and one sonoff basic for controlling light. 

Still doesn't answer my question that if I connect a temp probe to gpio14 and power/ground will tasmota recognise it ?

curzon01, 2018-02-26:
> if I connect a temp probe to gpio14 and power/ground will tasmota recognise it ?

Yes, but without build in rules.

> home assistant

is what I mentioned with external system