arendst/Tasmota #2027: Wemos D1 mini Switch/button NOT WORKING

hwarang098tkd, 2018-02-25:
module: wemos d1 mini
I tried to connect one button but i cant
I want to set a button to represent the relay1 . So , if i understand right    when i press the button1 the relay1 respond. I can understand also that if i set the specific Gpio to "button " or "switch" it configure as Input.... I notice that neither button or switch mode configured as output. 3.3v after set......
Am i right ? is it error on code?

Frogmore42, 2018-02-26:
Switch and button are working as designed.  You need to add a relay configuration for the output.

hwarang098tkd, 2018-02-26:
Sorry for my ENGLISH 
So I did this 
I connect for example on gpio d5 (set on Relay1) my relay and working fine and gpio d8 (button1) a classic button . Am I doing something wrong ? Is this button1 programmed to connect with relay1 ?

hwarang098tkd, 2018-02-26:
Sorry don’t this . I don’t Know what gone wrong but works fine now with the same wiring. The only thing is that I didn’t update wemo but I flashed it with the latest release .

closed 2018-02-26 by hwarang098tkd