arendst/Tasmota #2029: Sonoff Dual Push

alexmc1510, 2018-02-26:
Dear guys,

Firstly I would like to introduce myself. My name is Alejandro, from Spain. IT engineer and begineer but enthusiastic with sonoff and tasmota. I am deploying tasmota in my home automation system with Domoticz as brain. Yesterday I had and error while I was programming my roller shutters in a sonoff dual. I wanted to open and close my roller shutter when I pulse GPIO0 and GPIO9 buttons but I had a mistake and I program this 

SwitchMode<x> | 2 | Set switch mode to inverted FOLLOW (0 = On, 1 = Off)
-- | -- | --

instead of 

SwitchMode<x> | 3 | Set switch mode to FOLLOW (0 = On, 1 = Off)
-- | -- | --

After that, my sonoff restarted and doesn't connect to the network or appear as AP. I suppose, as program GPIO0 as inverted FOLLOW it restart in flash mode and I can't see it from the network. I have tried to restart with GPIO0 conected to ground but it doesn't work. I have not received FTDI programmer so I can't connect via Serial interface.

¿Any idea?

Thanks.

alexmc1510, 2018-02-27:
Hello guys,

Yesterday I was performing some tests with no results. Today I am going to restart conecting gpio0 to gnd directly, but I suppose that it will not work.

alexmc1510, 2018-03-09:
Hello guys,

Thank you for all the replies. I have solved the issue alone and I would like to share the solution. The problem was CFGHOLDER data. Because there is a backup system, it is necessary to change data stored in CFGHOLDER field for 1. After that, you have a completely cleaned tasmota system.
Correct me if I am wrong.

Regards,

ascillato, 2018-06-04:
Hi,

Sorry that nobody answer your issue before. There are lots of issues open and it is very difficult to follow everything.

Thanks a lot for sharing your solution. It is very appreciated. :+1:

closed 2018-06-04 by ascillato