arendst/Tasmota #2030: Relay triggered after Sonoff Basic restarts

PierreScerri, 2018-02-26:
Hi

I'm using a Sonoff Basic to control my garage door.  The relay switch has been isolated by cutting the PCB tracks.  The relay contacts are connected to position switch the is closed when the door opens.

The Sonoff is configured thus: 
  GPIO3 - 18 Relay2
  GPIO14 - 10Switch2
    PowerOnState 0 - prevents relay1 on after power outage - hopefully
    Pulsetime1 5
    Pulsetime2 0
    SwitchMode1 0
    SwitchMode2 2

This was working perfectly except that Hassio was not getting the state of the switch when Hassio restarts.  

I started playing around with the PowerRetain setting and now, whenever the Sonoff restarts after a power outage, I get:

00:00:00 Project sonoff Garage Door (Topic sonoff01, Fallback DVES_3C18C7, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP2 Study-TP-LINK_22AC in mode 11N as sonoff01-6343...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on sonoff01-6343.local with IP address 192.168.1.201
11:30:31 MQT: Attempting connection...
11:30:31 MQT: Connected
**11:30:31 MQT: tele/sonoff01/LWT = Online (retained)
11:30:31 MQT: cmnd/sonoff01/POWER = 
11:30:31 MQT: stat/sonoff01/RESULT = {"POWER1":"ON"}
11:30:31 MQT: stat/sonoff01/POWER1 = ON**
11:30:31 MQT: tele/sonoff01/INFO1 = {"Module":"Sonoff Basic","Version":"5.12.0","FallbackTopic":"DVES_3C18C7","GroupTopic":"sonoffs"}
11:30:31 MQT: tele/sonoff01/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff01-6343","IPAddress":"192.168.1.201"}
11:30:31 MQT: tele/sonoff01/INFO3 = {"RestartReason":"Software/System restart"}
11:30:31 MQT: homeassistant/switch/sonoff01_1/config =  (retained)
11:30:31 MQT: homeassistant/switch/sonoff01_2/config =  (retained)

The lines in bold are triggering the relay and opening the garage door.

Tried setting PowerRetain 0 then PowerOnState 0, but it does not seem to be working.

I am guessing that the culprit is the LWT line.  Why is it still there after I set PowerRetain 0?  How can this be removed?  Or have I completely misunderstood it?

Thank you.

wirelesssolution, 2018-03-07:
Hi , I got the same problem with my garage door. Wait some one to answer this issue. 

Thank you.

PierreScerri, 2018-03-07:
enter sleep 0 at the console.

wirelesssolution, 2018-03-08:
I enter 
11:40:53 CMD: sleep 0
11:40:53 MQT: stat/Garage_Door1/RESULT = {"Sleep":"0 (0)"}
and restart. it still ticker relay.
FYI I using Sonoff Relay, not external relay.

PierreScerri, 2018-03-08:
Can't remember exactly but I  believe that I ended up re-flashing  Tasmota when I couldn' get it to work. It's been working nicely since then.

closed 2018-06-04 by ascillato

nickidw, 2019-05-01:
There is a command PowerOnState that controls what the relay does after startup: https://github.com/arendst/Sonoff-Tasmota/wiki/PowerOnState-Configuration

PierreScerri, 2019-05-01:
Yes Thank you.

PowerOnState was set to 0 as per my post over a year ago.

The problem was the retained message.  Solved by re-installing MQTT