arendst/Tasmota #2031: Minimize RAM usage with webserver

stefanbode, 2018-02-26:
Hi @arendst, with the latest modifications my TLS version really struggle with the WEB interface. The issue is caused by the REPLACE functionality. Depending on the size of the page the memory is too low. Is there an option to replace in chunks of 1024 byte for example instead of the full page?

I do not know if this really solves the issue, but maybe an idea, if the overhead is reasonable. Without the TLS MQTT there is ~21kByte of RAM which is plenty enough. With the TLS this goes down to 8kByte. 

My current workaround is to disable MQTT temporary and reboot. I will also try, but your coding is much better than mine :-)

arendst, 2018-02-26:
Chunks won't work. Had that a few versions ago where most screens did not show. Apple hardware was very bad using chunked data. Chunk docs also mention minimum chunk sizes of 2k which makes management again very difficult.

As noted in the user_config.h I do not support TLS AND webserver at the same time. If it did work for you you were lucky. YMMV.

stefanbode, 2018-02-27:
I arendst. The idea was to chunk the replace on the String "page". So do a replace line by line instead of the complete String in the hope it will consume less memory. I made in my code some changes by introducing a page2 String and do the replace in a smaller string before adding it to page. For me it looks like there is some benefit to get at least "working" webpages (buttons are grey and not blue) unless there are more than 8300 bytes of heap. The replace in func ShowPage(); I skip if the memory is low. Header and footer are wrong, but the interface is useable.