arendst/Tasmota #2032: Sonoff Basic not working after power surge

daaf84, 2018-02-26:
Yesterday we had a tripped RCD. We had to examine where the problem was, so the power went of a couple of times. After that the Sonoff was not able to switch. I have a normal wall switch connected to GPIO14/GND. This wasn't working anymore... Also the button on the device itself didn't trip the relay.
There was no webserver when I connected to the IP. 
Today I reflashed it (hardwire) But no results. I think it's fried. 
When I press the button 4 times the AP comes up and the site does, but, not able to connect with 'admin' and no password. The page is also stretched out vertically, and has a long row of questionmarks in black squares. 

Anyone with similar problems?

Luckily, I ordered 5 new ones on Aliexpress. 
Could a new memory chip do the trick? And if so, which one should I pick?

LucaAghemo, 2018-02-26:
I had the same problem today, after some day of work. 
I put my sonoff last week inside a shade lamp, pulled 2 wires out for a switch connected to GPIO 14/GND, and it was working great. 
Today suddenly it went offline from MQTT, the GPIO14 switch works, but slowly and not constantly,  the sonoff is connected to the router but no web page is available.
Is it gone?

Jason2866, 2018-02-26:
Which firmware do you use? 5.12.0? 
How much free memory do you have?

LucaAghemo, 2018-02-27:
An update, I flashed again the firmware (5.11.1) and it works again. I think the problem has been caused (in my case) by a bad switch connected to GPIO14, giving fast on/off randomly. Fixed the switch, reflashed the sonoff, now everything seems to work. 
If it happen again (without the bad switch) I'll let you know.

closed 2018-06-04 by ascillato