arendst/Tasmota #2034: I'm getting RESULT = {"Command":"Unknown"} when sending the MQTT commands

diasarmando, 2018-02-26:
Hi
Im using the latest version 5.12.0C 

When I send the MQTT command the led blinks but not do the action.
For example I sent: 
`cmnd/boiler/POWER 1 `
the log says:
```
00:05:53 RSL: Group 0, Index 0, Command POWER , Data 
00:05:53 MQT: stat/boiler/RESULT = {"Command":"Unknown"}
```
The local buttons, web toggle and web console works fine.  

Is this an issue on the latest version?
Tks

closed 2018-02-27 by diasarmando

ferdydek, 2018-03-18:
I'm having exactly the same issue. Can you comment with what solved it or what were you (and I) doing wrong?

diasarmando, 2018-03-18:
It was my fault. When you send the message, you have a field for the command and other field for the data.
My error was putting the command and the data in the same field of the form.