arendst/Tasmota #2035: [HowTO] Expand Sonoff Basic to 4MB Flash

reloxx13, 2018-02-26:
Hey guys,

i want to share you a guide to expand the flash from 1MB to 4MB of a sonoff basic.

You can find it in the [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Howto:-Expand-Sonoff-Basic-to-4MB-Flash) :)

[image]

Oxyandy, 2018-02-26:
Yeah, and the real benefit to doing this ?

reloxx13, 2018-02-26:
- avoid flashing minimal firmware,

- use webserver and ssl

- be ready for future bigger firmwares with more features 👍

arendst, 2018-02-27:
Alas, iram is full already resulting in no more interrupt driven code. The amount of ram is very low which makes further expansion also impossible. TLS uses too much ram. Flash is not the issue.

Bottomline, upgrading flash is useless when you're out of ram and that's just the case with an ESP8266.

I have a ledunia with 32MB (yes that's 32MB) and still no reliable TLS as, you guessed it, no more ram available.

The only advantage of bigger flash is that, if it has inbuilt flash management, it moves the written flash pages around within it's chip leading to longer lifetime. But even then I still have to see an ESP8266 dead because of too many flash writes.