arendst/Tasmota #2036: Noob question for Switchtopic and how to connect between 2 tasmota devices?

kevinshane, 2018-02-27:
Hi so sorry for the noob question, but according to the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic) I can control the T3 switch1 to another tasmota device like a B1?

I have a sonoff T3 and a ailight lightbulb both flashed with tasmota.
I would like to have T3 first switch(POWER1) to toggle the light on and off, which means when I send on/off command to the lights, the T3 also gets the same msg and perform the same action(eg. light off, T3 first gang off, or switch T3 first gang off, the lights gets off as well) But how to do this via Switchtopic?

I try to define a command in the web console inside T3 based for the wiki
`Switchtopic cmnd/ailight/POWER`
nothing happen, T3 still acting like ...a normal T3
What's the correct way to setup a switchtopic to let T3 first gang controlling the light?(cmnd/ailight/POWER)?

Sorry for the very noob question, I need a babysit here, thank you guys so much ..

curzon01, 2018-02-27:
Set `SwitchTopic ailight` on your T3 - see [SwitchTopic mode 2 - Use case](https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic#use-case)

kevinshane, 2018-02-27:
@curzon01 Sorry man, I've try that, still nothing happened, still toggle the t3 itself but watching the light web console, nothing happen, here is the web log for the t3 switch
```
07:10:58 CMD: SwitchTopic ailight
07:10:58 MQT: stat/SonoffT2/RESULT = {"SwitchTopic":"ailight"}
07:11:39 MQT: stat/SonoffT2/RESULT = {"POWER1":"ON"}
07:11:39 MQT: stat/SonoffT2/POWER1 = ON
07:11:44 MQT: stat/SonoffT2/RESULT = {"POWER1":"OFF"}
07:11:44 MQT: stat/SonoffT2/POWER1 = OFF
07:11:46 MQT: stat/SonoffT2/RESULT = {"POWER2":"ON"}
07:11:46 MQT: stat/SonoffT2/POWER2 = ON
07:11:47 MQT: stat/SonoffT2/RESULT = {"POWER2":"OFF"}
07:11:47 MQT: stat/SonoffT2/POWER2 = OFF
```

curzon01, 2018-02-27:
sorry, SwitchTopic are for switches, your are using buttons:

`ButtonTopic ailight`

and reset switchtopic back to default

`SwitchTopic 0`

kevinshane, 2018-02-28:
@curzon01 Thanks man, that works, however this `ButtonTopic` command set all 3 gang to this mode, can I have only first gang to trigger `ButtonTopic` but leave the rest as original function? I setup the light listen only for POWER1, the light does turn on/off when touch the first gang(POWER1)... but the rest seems breaking(POWER2), here is the log from the t3 weblog:
```
04:04:29 CMD: ButtonTopic
04:04:30 MQT: stat/SonoffT2/RESULT = {"ButtonTopic":"ailight"}
04:04:59 CMD: SwitchTopic
04:04:59 MQT: stat/SonoffT2/RESULT = {"SwitchTopic":""}
04:08:08 MQT: cmnd/ailight/POWER1 = TOGGLE  // This is when the light turn on/off happen
04:08:10 MQT: cmnd/ailight/POWER1 = TOGGLE
04:08:12 MQT: cmnd/ailight/POWER1 = TOGGLE
04:08:16 MQT: cmnd/ailight/POWER2 = TOGGLE // This is when I touch gang2 happen
04:08:19 MQT: cmnd/ailight/POWER1 = TOGGLE
04:09:15 MQT: cmnd/ailight/POWER1 = TOGGLE
04:09:18 MQT: cmnd/ailight/POWER2 = TOGGLE // again gang2 seems listen to buttontopic too
04:09:19 MQT: cmnd/ailight/POWER2 = TOGGLE
04:09:21 MQT: cmnd/ailight/POWER1 = TOGGLE
```

kevinshane, 2018-02-28:
one more strange thing, when I type `cmnd/ailight/POWER1 ON` directly through web console, the first gang relay turned on(I can hear the click sound), it seems through web console it somehow, go back to original function
```
04:19:04 CMD: cmnd/ailight/POWER1 ON
04:19:04 MQT: stat/SonoffT2/RESULT = {"POWER1":"ON"}
04:19:04 MQT: stat/SonoffT2/POWER1 = ON
```

When touch the first gang(POWER1) physically, it seems listen to buttontopic to turn on/off the light
```
04:22:56 MQT: cmnd/ailight/POWER1 = TOGGLE
04:23:01 MQT: cmnd/ailight/POWER1 = TOGGLE
```

curzon01, 2018-02-28:
> ButtonTopic command set all 3 gang to this mode, can I have only first gang to trigger ButtonTopic but leave the rest as original function?

unfortunately this is not supported

> when I type `cmnd/ailight/POWER1 ON` directly through web consol

on web console the correct command is `POWER1 ON`, not `cmnd/ailight/POWER1 ON`.
The syntax you are using is a MQTT topic and unnecessary on console. On console simply use the commands itself, that's enough (it works because Tasmota uses only the part after the last slash `/` but `cmnd/xxxx/` on console is not correct).

> When touch the first gang(POWER1) physically, it seems listen to buttontopic to turn on/off the light

button does not _store_ a condition so TOGGLE will be used

kevinshane, 2018-02-28:
Thanks man, very clear explaination. So sad we can't use tasmota device to controlling another device.
I wish @arendst have a time to look at this:
a sonoff tasmota device like T1 3gang ---> 1gang controlling the ailight tasmota ---> 2gang controlling the b1 light ---> 3gang remains the original function to toggle non-smart lights

Jason2866, 2018-02-28:
Thats not the concept of mqtt controlled devices....

curzon01, 2018-03-01:
@Jason2866: I think that's a missunderstanding: kevinshane mention was that T1 'logical' controls directly another device, not directly at all. It needs a broker/server anyway, but the idea that T1 single buttons controls other devices without an extra system, why not?

Jason2866, 2018-03-01:
In my opinion this is not practicable. Too many variants and possibiltys...
The great benefit is that you can do everything you want with a HA and nothing is hard coded!

kevinshane, 2018-03-02:
@Jason2866 Hi, this is not HA or Openhab related, this is exactly the hardware setup issue.
when having many sonoff T1 and B1(or ailight) everywhere in house, we have to face the problem is that T1 acting like a **physical** relay to "cut off" the electrocircuit (buttontopic=0), in this case any tasmota device like B1 light will go completely offline if the wire comes from T1.

I follow curzon01's advise to setup the T1 into a buttontopic=ailight to directly turn T1 into a "send-mqtt-msg-only" device, which will not "cut off" the electrocircuit but keep the light alive.

However the problem is when setting T1 into a buttontopic=2 mode makes all gang(3gang version T1) into this mode, which is more or less useless for the rest of 2 gangs. SO, we have to leave the T1 "always ON" but controlling the on/off state using B1 POWER mqtt commands.(and it's painful)

What I trying to request a feature, is make the T1 one of the gang into buttontopic2 mode but leave the rest as original function, but looks like currently we don't get this feature, yet..

Sorry for my poor English, I hope I make myself clear, thanks ^^..

davidelang, 2018-03-02:
On Wed, 28 Feb 2018, kevinshane wrote:

> Thanks man, very clear explaination. So sad we can't use tasmota device to controlling another device.
> I wish @arendst have a time to look at this:
> a sonoff tasmota device like T1 3gang ---> 1gang [...]

you need to have the device send the the message to the mqtt server, and then 
have some software (typically some home automation software) decide what to do 
and send out the various commands to whatever devices need to be told to do 
something.

the ESP8266 just doesn't have the space to host the same complex logic that 
you can put on a raspberry pi running openhab or similar.

You are far from the first person to ask for one device to directly control 
others, but every person who has asked has a slightly different use case and the 
code that would work for one wouldn't work for others.

revk, 2019-04-04:
I'd like to see each button (or switch) configurable in terms of what MQTT topic it sends, with the default being the one internally to change the corresponding relay. That way I can have one button work another light, or a GroupTopic of more than one light even, without a separate controlling system on my network (I have a simple MQTT broker). Surely that is not too much to ask, and looks an awful lot like it should work from the documentation. When I try (v6.5.0) it returns a result ButtonTopic 0 whatever I sent it and does not work...