arendst/Tasmota #2037: T1 Touch Increasing Backlight Touch Brightness

jeylites, 2018-02-27:
I own two versions of Sonoff T1 UK -  the old version has got dual LED, it’s bright enough to see the button during OFF mode. However, the new ones only have one LED, when it’s off I could bearly see the button. Is there a way to increase the brightness?

stefan-hoehn, 2018-03-08:
I actually have the similar issue with the S20: In my case I feel that the brightness is too high (I have 2 LEDs) and I would love to lower the brightness. 

However, I doubt that there is a way to be able to dim the LEDs by CPU (it would mean that the output of the GPIO would not have to be a digital output but an analog and why would they do that...). The only way I can think of is to change the resistors in front of the LED or add an additional LED.

santibur06, 2020-07-25:
hey @jeylites, were you ever able to figure this out?