arendst/Tasmota #2038: RF-Bridge Not learning code / no Beep

nicecube, 2018-02-27:
Hi, i was using Tasmota from 2 month now with no issue, i have like 6 Sonoff Basic, 1 Dual and 2 S20 all working fine. I brough the RF-Bridge with the PIR MOtion Detector, i was not able to pair the device.

[image]

When i trigger the PIR i see a red dot on the RF-Bride but no MQTT was recived, i can see that my bride can send MQTT to my server because i receved a lot of info from the Bridge
[image]

When i tried to learn a RF frequency from the bridge i canot ear the Beep, also i see nothing in the console just : 
`05:01:08 CMD: rfkey1 2	
05:01:08 RSL: Group 0, Index 1, Command RFKEY, Data 2	
05:01:08 MQT: stat/RF-Bridge/RESULT = {"RfKey1":"Start learning"}`

Jason2866, 2018-02-27:
Why do you think that the RF Bridge didnt learned the code?
Last message is  RfKey1 has learned the code and saved it.
Nothing more or less you can expected from the RF Bridge.
What reaction should be done when a PIR Sensor "fires" has to be done
In the Home Automation systems which receives or better saying has to decode
the mqtt messages.

nicecube, 2018-02-27:
**Why do you think that the RF Bridge didnt learned the code?**
Because normaly im supposed to have something like this:
18:22:20 MQT: stat/sonoff433br_2/RESULT = {"RfKey16":"Start learning"}
18:22:20 SER: Received A0 55
18:22:27 SER: Received A3 27 74 01 40 03 D4 55 45 14 55
18:22:27 MQT: stat/sonoff433br_2/RESULT = {"RfKey16":"Learned"}

Normaly when my PIR trigger i am supposed to recieve a MQTT but nothing

Jason2866, 2018-02-27:
Ok. got it. Yes you are right. Some codes the RF Bridge is unable to learn.
I have this too. The problem is the extra ic which has the rf decoding job.
The firmware of this chip is untouched -> original.
I dont use the learning function it is not need when using a home automation system
Another benefit you are not limited to 16 keys

Jason2866, 2018-02-27:
Do you see a reaction on the web console of the RF Bridge when the PIR triggers?

nicecube, 2018-02-27:
No, i see nothing on the web console and no MQTT signal, this is the main issue, when i manually trigger de PIR i see a red dot on the brige so it recived the signal. I was not abble to par the PIR to the Bridge i do nothing so i dont know if the bridge and the PIR was connected the only clue is connected is the red dot.

btw sorry for my english im using google translate

Jason2866, 2018-02-27:
Google translate is sometimes better than my english...
There is no binding or on board function for items when using tasmota software.
To get things sorted out. Delete all learned keys. Now trigger your PIR.
If than is no output on the web console of the rf bridge something is generally wrong!
Do you have a simple rf 433 Mhz Transmitter/  Remote? If yes, do you have a rf received
message on the web console?

nicecube, 2018-02-27:
I ll try to remove all my rf key when im goin home, i ll also try to learn another RF frequency like my AC remote.

Jason2866, 2018-02-27:
AC remotes are often not working, not supported protocol. Just try it.
The PIR when buyed together as a bundle with the bridge must be working.

nicecube, 2018-02-27:
Okay, I also tried with my Roku controller but the bridge does not react. What I find strange is that the Bridge reacts well to the sensor because I see the red led light but I can not see anything in the console. Normally I should see that it has trigger and send me a topic MQTT

Jason2866, 2018-02-27:
Yes thats strange. This is the reason why i would reset everything and start from scratch

nicecube, 2018-03-04:
I flashed it again and i have the same issue, do you have another idea i can test

Jason2866, 2018-03-04:
I think the bride doesnt support the protocol of your remote.
In that case you out of luck...

nicecube, 2018-03-04:
Its not my remote, its the PIR Motion Detector that come with the kit, i ll recive my battery for the 4 boutons remote and the SW1 Sensor this week.