arendst/Tasmota #2039: KMC 70011 Module power calculation is wrong

3ricj, 2018-02-27:
I'm not sure why, but.. 

KMC 70011 Module

120 volts * 3.312 amps = 1348 watts.  

This must be new math . By my math, it should read: 

120 volts * 3.312 amps = 397.44 watts.  

Latest version of Sonoff Tasmota, Calibrated using the following values: 

VoltageCal 1770 | CurrentCal 1210

(confirmed with readings on watt meter).

3ricj, 2018-02-27:
[image]

3ricj, 2018-02-27:
Ok, now that I've read the fine manual... I'm even more confused.  Why are there separate PowerCal settings when it should be calculated based on the input voltage and current?

curzon01, 2018-02-27:
simplified `P = U * I` only for DC
on AC simplfied formular only for ohmig-only loads - the most loads are typically not ohmig-only

arendst, 2018-02-27:
KMC 70011?????????

guzzit3, 2018-03-08:
It only has one voltage and one current sensor but it can calculate when these occur during since wave cycle.Any displacement of the two is expressed as the power factor and this needs used to caculate the useful power that is being used. The device is reporting the power you are being charged for!!

zerog2k, 2018-03-17:
Assuming you have used the calibration steps here: https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Pow#calibration
I prefer the alternative method listed, as I have such an incandescent bulb to present a (nearly) pure resistive load, and I have also plugged it in to a watt-meter (e.g. Kill-A-Watt) to cross reference the actual values.

The measurement IC is HLW8012, and if you are interested in how the measurement and reporting is done, check out this blog entry:
http://tinkerman.cat/hlw8012-ic-new-sonoff-pow/