arendst/Tasmota #2040: Hassio and Sonoff s20 with tasmota

Steakschen, 2018-02-27:
Hi,
i got my 2x S20 flashed with tasmota and connectet als "light" with my hass.io

I cant figure out how retain option and the SwitchMode option working together.

I cant use the button on the s20. I switches the state for 1 sec and then gets overwritten by the hassio-web-state.
Whatever state i switch in the web works, but the button have not the "power/permission" to change the web-state.

Thanks

`light fernseher:
  - platform: mqtt
    name: "Fernsehen"
    state_topic: "stat/SonoffWohnzimmer/POWER"
    command_topic: "cmnd/SonoffWohnzimmer/POWER"
    availability_topic: "tele/SonoffWohnzimmer/LWT"
    qos: 0
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "online"
    payload_not_available: "offline"
    retain: true`
`

MQTT Host | m23.cloudmqtt.com
-- | --
13asdasd5
asdasda
SonoffWohnzimmer
SonoffWohnzimmer
sonoffs
cmnd/SonoffWohnzimmer/
`

vMeph, 2018-02-27:
On web console put PowerRetain 1?

Steakschen, 2018-03-06:
yes, doesnt change anything :-(

vMeph, 2018-03-06:
qos: 1 ?

Steakschen, 2018-03-06:
changed from 0 to 1, but same, web state overwrites local state within 1 second

Steakschen, 2018-03-09:
no, still isntantly switches back

Steakschen, 2018-03-14:
nothing works :-(