arendst/Tasmota #2041: Slow and reconnecting when connected to MQTT

Heroj04, 2018-02-27:
Using version 5.12.0 on a sonoff basic
When not using MQTT (disabled, incorrect details, broker not on) tasmota works fine through the web interface.
however as soon as it starts to use MQTT it appears to be very slow, not reacting to MQTT messages, webpages loading extremely slow.
I ran just a ping to it continuously and I can see while not using MQTT i get ping returns consistently with 1-3ms but once MQTT is connected ping returns seem to stop consistently.
```
Pinging 192.168.1.183 with 32 bytes of data:
Reply from 192.168.1.183: bytes=32 time=3ms TTL=128
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.1.183: bytes=32 time=1ms TTL=128
Reply from 192.168.1.183: bytes=32 time=14ms TTL=128
Request timed out.
Request timed out.
Reply from 192.168.1.183: bytes=32 time=4ms TTL=128
Reply from 192.168.1.183: bytes=32 time=1ms TTL=128
Reply from 192.168.1.183: bytes=32 time=2ms TTL=128
Reply from 192.168.1.183: bytes=32 time=15ms TTL=128
```

I checked the weblog and I can see it trying to reconnect constantly:
```
13:26:14 MQT: Attempting connection...
13:26:16 MQT: Connected
13:26:16 MQT: tele/sonoff/LWT = Online (retained)
13:26:16 MQT: cmnd/sonoff/POWER = 
13:27:34 MQT: Attempting connection...
13:27:36 MQT: Connected
13:27:36 MQT: tele/sonoff/LWT = Online (retained)
13:27:36 MQT: cmnd/sonoff/POWER = 
13:27:36 DNS: Initialized
13:27:45 WIF: Connecting to AP1 Footner in mode 11N as sonoff-0618...
13:27:50 WIF: Connected
13:27:50 DNS: Initialized
13:27:51 MQT: Attempting connection...
13:27:52 MQT: Connected
13:27:52 MQT: tele/sonoff/LWT = Online (retained)
13:27:52 MQT: cmnd/sonoff/POWER = 
13:28:04 DNS: Initialized
13:28:04 MQT: Attempting connection...
13:28:06 MQT: Connected
13:28:06 MQT: tele/sonoff/LWT = Online (retained)
13:28:06 MQT: cmnd/sonoff/POWER = 
13:29:00 DNS: Initialized
13:29:00 MQT: Attempting connection...
13:29:02 MQT: Connected
13:29:02 MQT: tele/sonoff/LWT = Online (retained)
13:29:02 MQT: cmnd/sonoff/POWER = 
13:29:24 MQT: Attempting connection...
```

I've tried using Mosquitto on my Home Assistant RPI3, the built in MQTT broker from HASS and CloudMQTT, all produce the same results.

any idea whats going on?

Frogmore42, 2018-02-27:
It is having trouble connecting.  There are lots of reasons this might be.  Since it seems to work when not using MQTT, which core version are you using.  One of the status pages has this.  You are probably using 2.4.0, which some people have noticed causes issues, while 2.3.0 seems to work fine.

Heroj04, 2018-02-28:
I am using core 2.4.0
`Core/SDK Version | 2_4_0/2.1.0(deb1901)`
How would I go about downgrading?

alton76, 2018-02-28:
There’s seem to be issues with ESP V2.4. To downgrade is the same way you
installed V2.4 in Boards Managers. There will be a tab to remove versions
and install.

So Arduino - Board Managers - Search ESP- Select Version 2.4 - Remove -
Select Version 2.3 - Install

Don’t forget to copy Tasmota prerequisites into the ESP V2.3 directory.

Heroj04, 2018-02-28:
Thank you guys so much for the help.
Downgrading to 2.3 has solved my issues.
It's now working seamlessly with and without MQTT.
wonderful

closed 2018-02-28 by Heroj04