arendst/Tasmota #2042: RF Bridge - WiFi Loop/Not Connecting

lanky8804, 2018-02-27:
So have flashed Sonoff RF Bridge using Arduino, just changed the WiFi SSID and password in the config file as per instructions. However when i unplug flick switch and plug back in am getting this in the serial monitor over and over again. Cannot connect to the RF bridge

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_8266C9, GroupTopic sonoffs) Version 5.11.1
00:00:00 WIF: Connecting to AP1 TheFluffyCloud in mode 11N as sonoff-1737...
00:00:04 WIF: Connected

Exception (2):
epc1=0x3ffeeb74 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffeeb74 depc=0x00000000

ctx: cont 
sp: 3fff2c90 end: 3fff30d0 offset: 01a0

>>>stack>>>
3fff2e30:  0000002e 3fff35d0 3fff3c7c 401043d1  
3fff2e40:  3fff410c 3ffe8320 00000009 00000064  
3fff2e50:  4026e3a2 00000008 3fff0c20 3fff3605  
3fff2e60:  3fff35d0 3fff35d0 3fff410c 4010453d  
3fff2e70:  402365ab 3fff410c 3fff4152 402365b4  
3fff2e80:  00000008 3fff2eb0 3fff2e90 40228638 <
3fff2e90:  3fff1fac 0000014f 0000014f 3fff4160  
3fff2ea0:  3fff410c 3fff40f8 3fff35d0 4023c68d  
3fff2eb0:  005e0001 0000fb00 3fff428c 40100670  
3fff2ec0:  00000018 3fff40f8 3fff410c 4023d228  
3fff2ed0:  00000020 3fff2f3c 3fff2f3c 00000000  
3fff2ee0:  3fff2f54 00000001 00000000 00000002  
3fff2ef0:  3fff3484 00000004 00000002 4023d8ac  
3fff2f00:  40260577 00000002 3fff2f48 3fff40f4  
3fff2f10:  00000000 3fff410c 3fff4178 4023d268  
3fff2f20:  3fff35d0 3fff2f50 00000004 3fff40f4  
3fff2f30:  00000016 3fff35d0 3fff40f4 4023c8e5  
3fff2f40:  3fff35d0 3fff2f50 00000004 40236610  
3fff2f50:  00000494 0104a8c0 3fff387c 4023664c  
3fff2f60:  3fff40f8 3fff35d0 3fff40ac 4023c9ba  
3fff2f70:  3fff408c 3fff1df4 3fff3020 00000020  
3fff2f80:  3fff2fec 3fff35d0 3fff40f4 4023cb93  
3fff2f90:  3fff2fec 3ffea4b4 3fff35d0 4023cc1f  
3fff2fa0:  3fff4038 00000001 3fff3020 00000001  
3fff2fb0:  3fff1e30 3fff1df0 3fff1e30 40221f62  
3fff2fc0:  00000000 3fff1df0 3fff3010 40100690  
3fff2fd0:  3fff1fac 00000142 00000000 40229b98  
3fff2fe0:  00000000 3fff1df0 3fff1e5c fb0000e0  
3fff2ff0:  00000000 3fff1df0 3fff1e30 0000006e  
3fff3000:  00000000 3fff1df0 3fff1e30 402234b2  
3fff3010:  3fff41f4 3fff3040 40221c58 40222084  
3fff3020:  00000000 00000000 0101a8c0 0000006e  
3fff3030:  3ffe86f8 3fff1df0 3fff11cd 4021532b  
3fff3040:  3ffe9f00 aa01a8c0 3fff176c 40106e61  
3fff3050:  3fff3c7c 3fff1710 3fff16c5 0000006e  
3fff3060:  3ffe86f8 3fff1710 3fff1700 402157dd  
3fff3070:  3fff176c 3fff110e 3fff20b0 00000000  
3fff3080:  3fffdad0 3fff20a8 40229d4c 3fff20b0  
3fff3090:  3fff1268 00000062 3fff20a1 3fff20a8  
3fff30a0:  3fffdad0 00000000 3fff176c 4021a92a  
3fff30b0:  3fffdad0 00000000 3fff20a1 40229d98  
3fff30c0:  feefeffe feefeffe 3fff20b0 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

Jason2866, 2018-02-27:
I think flashing failed. What you describe is a typical behaviour for that

lanky8804, 2018-02-27:
Flashing completed with no error messages displayed

Jason2866, 2018-02-27:
If flash fails there is not always a error message... My Led wifi controller only worked after 3rd flash.
Do you have compiled your own version with your settings in user_config.h?
Using a general version sometimes is not working.

lanky8804, 2018-02-28:
For anyone else with this issue, solution for me was:
Arduino v1.8.5
Tasmota v5.12.0
ESP8266 Boards Manager v2.4.0

Followed the tutorial as per wiki but changed IwIP to v1.4 Prebuilt.

Please see screenshot below.

[image]

closed 2018-02-28 by lanky8804