arendst/Tasmota #2043: Invert logic on Relay

choriz80, 2018-02-27:
Hi Arendst,

I just start using you soft with SonOff, and many thanks for your works it 's amazing  ! and work perfectly.

I just an issue for invert relay logic (when i have 220v on relay output, i would have switch off status on domticz), i tried to use SwitchMode 0 to 7 with command line but nothing change (i rebooted each time).

Hope you could help me

Many thanks and regards

Jason2866, 2018-02-27:
SwitchMode is not for that. Thats for behaviour if you connect a extra switch to the module.
If you want inverted logic you have to change the GPIO definitions.
Select module Generic and do the config you want/need.

choriz80, 2018-02-28:
Hi Jason,
Thanks for your answers, i will try to increase my experience to build as generic module.
However it could be a very good option to have as command line :)
Especially for French pilot heating who stop heating when electric wire(pilot) is on. 

Regards

tbart, 2019-08-26:
Click on Configuration-Configure Template, choose your hardware in "Based on", give your template some name and reconfigure the relay you want to use inverted logic on to relay<x>i instead of the default relay<x>.
Click Save.
Click on Configuration-Configure Module, choose your newly generated template.
Profit.

Brackbouille, 2019-10-12:
Hi tbart,

i tried you re solution, that work good thanks. 
I had already solved the issue by custom firmware simply with _INV in the code

Regards

SzilagyiR, 2021-02-06:
> Click on Configuration-Configure Template, choose your hardware in "Based on", give your template some name and reconfigure the relay you want to use inverted logic on to relayi instead of the default relay.
> Click Save.
> Click on Configuration-Configure Module, choose your newly generated template.
> Profit.

Thanks a lot, you solved my problem :)