arendst/Tasmota #2044: Switches Laggy when Wifi Router is turned off.

DNA99, 2018-02-27:
Hi, I have quite a number of sonoff basics on Tasmota ver 5.12.0 and I have GPIO14 set as Switch1.
All these Sonoffs are connected a Rasberry Pi running on HASSio. Everything runs great when the sonoffs are connected to Home Assistant. 

There are 2 things which I have noticed with the Sonoffs:
1. It is rather difficult to connect directly to the sonoffs after they have been turned on for some time. For most of them I have to switch off the main switch from the circuit breaker to reset the sonoffs before I can connect to them.
2. When the home router is turned off or if raspberry pi running the home assistant is switched off. it is rather laggy when I try to switch on/off the lights using the flip switches that are connected to the sonoff on GPIO14.

Is there anything I can do to rectify these 2 issues?

jeylites, 2018-02-27:
For question two, it got to do with not connecting to MQTT broker which is hosted on Raspberry Pi I assume...

DNA99, 2018-02-27:
@jeylites Yes. seems that the sonoff is trying hard to reconnect to the MQTT server and isn't able to respond promptly to the GPIO inputs. Is this a know issue?

jeylites, 2018-02-27:
@DNA99 yes it is.