arendst/Tasmota #2045: Sonoff Touch, delay between toggle

azor1an, 2018-02-28:
Hi, I'm trying to configure a Sonoff Touch so it behaves like the following:
- toggle state immediate when I touch the button
- minimal wait time between accepting a new touch input 

I've done "SetOption13 1", which makes the relay toggle almost instant after I touch the button, however this only happens reliably if it has been been a few seconds from the last button toggle. If I touch the button too quickly after the last button press it most of the time ignores that touch.

Is there any other config option that I can set to minimize this delay between accepting button touches?
thanks.

ionciubotaru, 2018-02-28:
Check your mqtt server, the strength of you wireless signal - a poor mqtt connection can cause those delays

azor1an, 2018-03-01:
you are right @ionciubotaru. I moved the MQTT server closer to the switch and it now reacts very quickly.

closed 2018-03-01 by azor1an