arendst/Tasmota #2046: Flashing Sonoff POW works, but not S20

Hekoweb, 2018-02-28:
Hi,

I tried to flash my Sonoff S20 without success, these messages came up:

**_Looking for upload port...
Auto-detected: COM5
Uploading .pioenvs\sonoff-DE\firmware.bin

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error -1
 [ERROR] Took 35.02 seconds_**
 

Building went fine.
The COM-Port was correctly recognized as COM5. I pressed the button to start upload mode. 

I tried it with another new S20, same problem.

Than I tried it with the same firmware and the same adapter on a Sonoff POW --> And that worked!

Is there something different with the S20? Any ideas?

curzon01, 2018-02-28:
interchange TX/RX lines

Hekoweb, 2018-02-28:
I had two issues, a cold solder joint and ones incorrect TX/RX connections.
Now it works.

closed 2018-02-28 by Hekoweb