arendst/Tasmota #2048: Wemos D1 mini

kmashal, 2018-02-28:
I downloaded the code to Wemos D1 mini , under configuration I do not see Wemos D1 mini as an option , also when I use the general configuration most of sensors do not display on main page  .

jeylites, 2018-02-28:
It falls under “generic”

kmashal, 2018-02-28:
when I configure it as generic non of the sensors works ? like DHT11 that I tried

reloxx13, 2018-02-28:
please provide more infos.

Jason2866, 2018-02-28:
Correct pin?  Correct definition? Which sensor? 
Wemos d1 mini works perfect with sensors if the setup is correct....
You have to give concrete informations.  Without informations nobody can and will help you....

kmashal, 2018-02-28:
I selected D3 pin to be my DHT11 input  when the board restarted after configuration change  it did not display any readings  ... when i assign Relay to D3 it works fine but not DHT11... I have the latest code downloaded to the board

kmashal, 2018-02-28:
correct the resistor is installed between vcc and data pin of the dht11

kmashal, 2018-03-01:
Do you have steps for programming Wemos D1 mini? I just want to make sure I did that correctly

jhlweb, 2018-03-01:
As what did you set the D3 pin in the configuration page?
[https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-Configuration ](url)

kmashal, 2018-03-01:
I set D3 to DHT11 and according to the documentation it should show temp values above Main Menu but nothing show, I suspect my code build have an issue any guidance on how to build a code for Wemos D1 mini  environment ?

Jason2866, 2018-03-02:
The wiki describes it very good
https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisite

kmashal, 2018-03-02:
I read all what is in your wiki , there is nothing about Wemos D1 mini setup , I used this setup steps
https://github.com/tIsGoud/Tasmota-on-WEMOS

Jason2866, 2018-03-02:
A very detailed setup guide! Did the Build process with Atom.io finished without errors?
Did you flashed the Wemos with Atom.io "Upload"? Finished successful?
If every answer is yes i think the sensor is defect.
In the new Version of Tasmota you have to choose Modul "Generic" there is no more modul Wemos!

RaymondMouthaan, 2018-03-12:
Recently I added [WeMos](https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos%20D1%20Mini) to Tasmota Wiki. 

Hope this works for you.

jeylites, 2018-03-12:
@RaymondMouthaan nice write-up. Thank you!

kmashal, 2018-03-13:
Thanks a lot :)

RaymondMouthaan, 2018-03-13:
@kmashal @jeylites You are welcome :) Please close this issue if it works for you.

closed 2018-06-02 by ascillato