arendst/Tasmota #2053: MQTT reconnection relay status switched to ON, toggled or OFF.

jnherm, 2018-03-01:
Why is it that after MQTT broker reconnection, my relay (Touch, Basic) were sometimes turn ON, OFF or Toggled. What will I set so that it won't do anything on the relay after MQTT reconnection? Thank you!

jnherm, 2018-03-01:
This is always the log after re-connection
21:07:28 MQT: Connected
21:07:28 MQT: tele/sonoff/02/LWT = Online (retained)
21:07:28 MQT: cmnd/sonoff/02/POWER = 
21:07:28 MQT: stat/sonoff/02/RESULT = {"POWER":"ON"}
21:07:28 MQT: stat/sonoff/02/POWER = ON

davidelang, 2018-03-02:
see the wiki page on poweronstate, especially the section on retained messages. 
It sounds like your mqtt server has some retained messages that it's sending to 
you

if you increase the logging level, you will see exactly where the command to 
turn on is being generated

jnherm, 2018-03-05:
Ok thank you but my setting on Power On State is #3 which will restore previous state before power off. And I don't think this is an issue about PowerOnState since the device did not reset nor power cycled during MQTT re-connection. But I will increase the level of my log hoping to find the culprit. I also did study the source code and the examples of re-connection, I found out that after re-connection the code did something on the state of the relay.