arendst/Tasmota #2054: Rf Bridge learning key  failure

JPSINHA, 2018-03-01:
I flashed with Sonof Tasmota 5.12.0d. But unable to use as Rfkey learning problem.

00:01:15 CMD: rfkey1 2
00:01:15 RSL: stat/sonoff/RESULT = {"RfKey1":"Start learning"}
00:01:26 MQT: Attempting connection...
00:01:27 DNS: Query done. MQTT services found 0
00:01:27 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
00:01:37 MQT: Attempting connection...
00:01:38 DNS: Query done. MQTT services found 0
00:01:38 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
00:01:49 MQT: Attempting connection...
00:01:50 DNS: Query done. MQTT services found 0
00:01:50 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
00:02:00 RSL: tele/sonoff/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:01:52"}
00:02:01 MQT: Attempting connection...
00:02:02 DNS: Query done. MQTT services found 0
00:02:02 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
00:02:13 MQT: Attempting connection...
00:02:14 DNS: Query done. MQTT services found 0
00:02:14 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
00:02:15 RSL: stat/sonoff/RESULT = {"RfKey1":"Learn failed"}
00:02:25 MQT: Attempting connection...

kovaccs, 2018-03-01:
Maybe look at the # 2023 closed issues.

JPSINHA, 2018-03-02:
Program Version | 5.12.0d
-- | --
2018-03-01T11:30:14
2_4_0/2.1.0(deb1901)
0T00:05:27
30 at FA000
18
Software/System restart
 
SJPS (60%)
sonoff-5301
192.168.43.16
192.168.43.1
255.255.255.0
192.168.43.1
DC:4F:22:82:D4:B5
 
 
1883
DVES_82D4B5
DVES_USER
sonoff
sonoffs
cmnd/sonoff/
 
Belkin WeMo
Enabled
Web Server
 
8574133
1327185
1024kB
1024kB
496kB
504kB
22kB

Not learning Rfkey
03:43:19 CMD: rfkey1 2
03:43:19 RSL: stat/sonoff/RESULT = {"RfKey1":"Start learning"}
03:43:20 UPP: Multicast (re)joined
03:43:30 MQT: Attempting connection...
03:43:31 DNS: Query done. MQTT services found 0
03:43:31 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
03:43:32 UPP: Multicast (re)joined
03:43:42 MQT: Attempting connection...
03:43:43 DNS: Query done. MQTT services found 0
03:43:43 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
03:43:44 UPP: Multicast (re)joined
03:43:54 MQT: Attempting connection...
03:43:55 DNS: Query done. MQTT services found 0
03:43:55 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
03:43:56 UPP: Multicast (re)joined
03:44:06 MQT: Attempting connection...
03:44:07 DNS: Query done. MQTT services found 0
03:44:07 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
03:44:08 UPP: Multicast (re)joined
03:44:18 MQT: Attempting connection...
03:44:19 DNS: Query done. MQTT services found 0
03:44:19 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
03:44:19 RSL: stat/sonoff/RESULT = {"RfKey1":"Learn failed"}
03:44:20 UPP: Multicast (re)joined

How can I come out from the problem?

kovaccs, 2018-03-02:
I think I have tow news one bad one good.
Bad: unfortunately t bridge probabily won't learn your remote codes. Can't control it throug web interface buttons. :-(
Good: There's a possibility to control your remote device trough MQTT command on Backlog topic.
--
To do:  Open consol on tasmota web. Push the sutable button on remote. Read the consol message like this:    
13:03:32 MQT: tele/sonoff433br_2/RESULT = {"RfReceived":{"Sync":13080,"Low":420,"High":1260,"Data":"371A34","RfKey":"None"}}
Maybe have to push many times if the result different (this is a game :-(   )

Construct multiply payload that you will send MQTT Backlog topic 
Like this:
topic: cmnd/<yourdevice>/Backlog
payload:  RfSync 13080; RfLow 420; RfHigh 1260; RfCode #371A34
If your remote controlled device doesn't reply, chose another reading among consol messages.

That's all! This helped me with the same problem.

Jason2866, 2018-03-02:
@JPSINHA First step. You have to setup a mqtt broker that is working.
The log says NO mqtt service reachable Before that everything is useless.

JPSINHA, 2018-03-02:
How to set mqtt broker? Kindly guide me.

Jason2866, 2018-03-02:
For a RasPi you can get a ready Diskimage for the the Homeautomation System OpenHab.
In this Setup all is included. Infos here:
https://docs.openhab.org/installation/index.html
Download SD Card image
https://bintray.com/openhab/bin/download_file?file_path=openhabianpi-raspbian-201712171718-git30faec3-crccc053823.img.xz
Be warned you have to read some stuff.

kovaccs, 2018-03-02:
If you don't want to use OpenHub, this is an general mosquitto install step by step instruction:
http://www.instructables.com/id/Installing-MQTT-BrokerMosquitto-on-Raspberry-Pi/
(Sorry I haven't known you havent got mosquitto installed :-O  )

JPSINHA, 2018-03-03:
RF KEY LEARNING can't be done through web

johoberm, 2018-03-03:
There isnt really any reason for laerning Rf keys. Everything you need is sent via mqtt.
If you use TASMOTA Firmware you have to use mqtt. So if you dont like that TASMOTA isnt the right decision for you.

kovaccs, 2018-03-03:
@JPSINHA: I think you haven't read my post before!!!

JPSINHA, 2018-03-04:
Can anybody guide me to start MQTT through Smartphone App MQTT dashboard for learing Rf Key of sonoff Bridge Tasmota 5.12d