arendst/Tasmota #2056: Looking for Sonoff Tasmota beta testers for Yeti app

luisfpinto, 2018-03-01:
Hello! 

I've decided to implement Sonoff into [Yeti](https://getyeti.co), a mobile app (Android & iOS) to control smart home devices.  I've started adding control for switches and I hope I can add sensors soon. 
Before releasing the switch control to production I would like some of you to test the App to see if the sonoff tasmota implementation it's working or not. This would be really helpful

I'll be opening the beta soon. You can join the list right [here](https://docs.google.com/forms/d/e/1FAIpQLScXKPY_i4d5b4Fs5FKz0w2bUVf-sHr_8GI6orVP905CrrkGXg/viewform?usp=sf_link)

th3magpi3, 2018-03-01:
Hey man, might want to make this public...

[image]

luisfpinto, 2018-03-02:
> @ th3magpi3 Hey man, might want to make this public...

I'm so sorry, you're right!! I think you should have access to it now.

mizrachiran, 2018-03-14:
Hi @luisfpinto 
How it's going with this integration? I would love to see tasmota in Yeti.
do you need help with testing \ anything else?

luisfpinto, 2018-03-14:
Hey @mizrachiran Yes for sure! We are right now in the beta testing process. Could you please introduce your email in the form? Remember to introduce the same email you use in the Play/App Store! 

Thank you so much!

mizrachiran, 2018-03-14:
I filled the form with the same email.
should I re-download the app on my phone?

luisfpinto, 2018-03-14:
You've received an email with the invitation! 👍

mizrachiran, 2018-03-14:
Thanks Luis.
it seems like the app doesn't auto discover my tasmota devices.
did I miss anything?

Stevehans, 2018-03-30:
Will yeti support the on board timers recently implemented in tasmota?

luisfpinto, 2018-04-02:
Hey! I'll take a look at the documentation and see if I can include it in the next release!

luisfpinto, 2018-04-02:
Hey @Stevehans, 

I will postpone this implementation as this can already be done with Routines inside Yeti!
Let me know if you have any issue with this functionality inside the app!

ascillato, 2018-04-02:
Hi, Yeti can find my tasmota devices and work vey well. But the timers are running on the phone. If I go away from home, my timers are not going to work. Can be added an option to configure the Tasmota Timers that already have the firmware?
thanks

Great App!!!

Stevehans, 2018-04-03:
Problem with routines is that if there is an issue with the cloud service (sonoff/ewelink issues being a classic example) then the central heating or whatever stops.

Having the ability to control the timers is a must I think.

luisfpinto, 2018-04-03:
Right now timers are controlled by Yeti locally through routines so we are not using any cloud service @Stevehans.

As @ascillato says, you're away timers won't work because Yeti is not on the same network. As an immediately solution if you have a tablet you can install Yeti on it and leave it at home. 

However, I will try to add timers control for the next releases ;) We need to do some more changes before releasing this.

closed 2018-06-14 by luisfpinto