arendst/Tasmota #2057: How to set the SwitchMode for the second Button on a Sonoff Dual R2

JueBag, 2018-03-01:
I'm struggeling with the implementation of external switches.
I have connected both Buttons of  a sonoff Dual R2 to Pushbuttons.
By using the command "SwitchMode  3" I get the report "stat/sonoff_dual/RESULT = {"SwitchMode1":3}"

Using this setting Button0 is acting as expected as an Pushbutton,  Button1 however does not!

Which syntax is needed to change the SwitchMode of  Button1?

I have set GPIO0 to "09 Switch 1" and GPIO9 to "10 Switch 2"

JueBag, 2018-03-01:
Found the solution:
In order to set the mode for Button0 use: 
`SwitchMode1 3`
for Button1 use:
`SwitchMode2 3`

closed 2018-03-01 by JueBag

davidelang, 2018-03-03:
switchmode only works on switches, not on buttons.

JueBag, 2018-03-03:
Sorry for using the wrong wording. 
"Button0" is the label of the pin which is connected to GROUND in order to perform the desired action of the switch.