arendst/Tasmota #2059: Please Note...

webtop, 2018-03-01:
This is not an issue with the code, but to highlight a point. I noticed that a lot of the “issues” here are not really issues and that they are just requests for help. After speaking with Theo, we decided to set up a forum dedicated to Tasmota, to separate the real issues from help, and discussions.

The forum has no content yet, and the site is not fully ready, but please feel free to sign up and start discussing topics.

URL: http://forum.digital-pig.com

We will be looking for contributions for articles, and also for moderators if you have expertise in a particular area.

Hopefully this will spread the great work that Theo and other contributors are doing, and drive a community to help each other in general.

MatzeJoerling, 2018-03-01:
could you pls rewrite to https://forum.digital-pig.com/

joopy, 2018-03-01:
Good to hear! Let's clean up the open 'issues'

roguestreak, 2018-03-02:
As per MatzeJoerling's comments, I think we'd first need confimation from Theo himself, and links off the wiki to make it official.  And https of course.
The wiki says the sonoffusers group is the support, but it seems that this "Issues" page is where everyone goes, and very few visit/reply on sonoffusers.

arendst, 2018-03-02:
Considering the amount of non code-relevant issues here I had a request from @webtop to facilitate a forum website.

As I know there is the google forum which is non-indexed I thought this could be a better solution so I appreciated his request. Although the link sounds a bit strange I do hope this forum will take the load off Github so I can start prioritizing issues much better to add/fix Tasmota.

I own sidweb.nl but I do not want to manage the forum so that's how we came up with this solution.

So I hereby announce the Tasmota forum at https://forum.digital-pig.com/ for open ;-)

webtop, 2018-03-02:
Thanks Theo. 

I will get an SSL cert set up this weekend, and also start adding links to the wiki and code. I'll also write some scripts to pull metrics from github for display in the forum.

As for the forum name, I had that available so used it, nothing more. Once the site is up-an-running we can create a more user-friendly name, and get a URL that makes sense. 

Again, anyone who thinks they have expert knowledge in any of the forum areas, and a little time to spare, please message me and I'll set you up as a moderator.

webtop, 2018-03-02:
SSL has now been implemented

MatzeJoerling, 2018-03-03:
Hi, 
SSL was already set up as i said rewrite it! what i want to say was that you should rewrite http requests to https! Nobody should (even accidentally) enter his/her credentials over http connection.

i didnt say Theo should run the forum, a cname forum.sidweb.nl  to 162.144.107.53 is all that is needed.

If you want to force users to the forum (in my opinion the ony way that works) is to:

- setup the forum the RIGHT way
- make this thread sticky at top of list
- close alle further cases here with "not relevant" (if not a bug, enhancement..)
- if possible, export all github issues to forum and make them invisible here

For running the forum:
- Dont make too much topics! (2 are enough, Tasmota / OffTopic Stuff)
- To many topics divide the good users on to much topics!
- if you need some domain tasmota.<whatevertdl> i will register a cheap one

webtop, 2018-03-03:
Hi Matze,

The SSL was not already there, as I hadn't purchased it until I mentioned on here. I was waiting for that to clear before setting up redirection, otherwise redirection is pointless. Redirection has now been set up.

As for the topics, it is important to have base topic groups for subscription purposes. By setting up a topic for Domoticz for example, allow users who use Domoticz to subscribe to just that topic and not have to deal with all the detritus from other topics they have no interest in. Hence the current topics, and to be honest, there aren't a lot.

It's not just a forum, it's a community project.

As to your other points, they are outside of my control.

webtop, 2018-03-03:
All users who registered and have not received an activation email...I have disabled email activation for now and activated all your account. Please log in when ready.

tyjtyj, 2018-03-04:
Do consider social login for registration/login plugin. Less one username/password to remember. 

Probably lock down to github only 

http://custom.simplemachines.org/mods/index.php?mod=3580

webtop, 2018-03-05:
@tyjtyj Good idea, I’ll get that implemented

tobox, 2018-03-05:
I personally like the idea to start a forum for tasmota, because many issues raised at github (including mine) were not really issues. 

My 2 cents:

* Don't create any topics, or just very few. Many questions touch more than one topic and then it is difficult to pick the correct one. Most of the home automation systems have their own forums, and communication is mainly handled through MQTT.

* Use https://www.discourse.org/ as a forum software. It so so much better than all others. As soon as users start putting their local setup infos into the signatures, the forum search resulst from google are often pointless.

emontnemery, 2018-03-28:
What happened to the forum, there's really a **ridiculous** number of issues in this project now...
@arendst How about evaluating this service: https://gitq.com/ ?

ascillato2, 2018-08-18:
Added Link in the wiki and in the issues templates. Thanks

closed 2018-08-18 by ascillato2

webtop, 2018-08-25:
I'm abandoning the idea of hosting a forum, because while there may be some requests for it in this thread, there was no real interest in it outside of this thread. People joined and then nobody bothered to post.

Hopefully gitq.com works out

ascillato, 2018-08-25:
Hi @webtop 

Sorry to see that you are closing the forum. Was a really great idea.

jhuizingh, 2018-08-28:
Well, I just read the first post and posted in the forum before seeing that it looks like it's not going to stay around. Any suggestions on where to post for help?

ascillato, 2018-08-28:
Taken from https://github.com/arendst/Sonoff-Tasmota#support-information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.