arendst/Tasmota #2060: ESP8266 accessed through slow internet connections: gets into strange mode

tomgregg, 2018-03-01:
Hi,
First, this firmware is terrific!!  I'm trying to read a few sensors and control a few relays from the internet, so it's the perfect setup.

I have three separate ESP8266's running in slightly different environments, all running Sonoff-Tasmota 5.12.0 code.  All are accessed over the WAN using DDNS for domain name assignment.  Sometimes, but not always, when I access these modules through a WiFi hotspot, or through public internet access, the modules are either unresponsive or get into a mode where they return what looks like a combination of raw http text and a few 'buttons.'  When I get back to my home where I have a more direct connection, they still return this combination of raw http text and a few 'buttons.'   However, the 'RESTART' button has always been visible, and when I push it, the modules restart to their typical 'good' state.  I have a few screenshots showing this behavior.  The only thing I can think that causes this is that it only happens with 'slower' connections.  It may also be something to do with more publically available internet connections, but I can't be sure.  I'm using two different android chrome browsers  (a smart phone and a tablet) and a windows machine.

Any ideas?  Thank you

[Sonoff-Tasmota fail 20180228.pdf](https://github.com/arendst/Sonoff-Tasmota/files/1772715/Sonoff-Tasmota.fail.20180228.pdf)

dYna77, 2018-03-02:
I have no solution, but i can confirm this beahviour.
When accessing the sonoff from outside the LAN i see exactly this.

reloxx13, 2018-03-02:
Did you use 2.3.0 or 2.4.0 ?

Their will be alot of fixes in 2.4.1, i suggest just to wait for 2.4.1 if you are on 2.4.0 and/or try 2.3.0.

tomgregg, 2018-03-02:
Hi,  This is what I'm using (2.3.0).  I'll wait for 2.4.1.  Thank you!
#define VERSION                0x050C0000   // 5.12.0

// Location specific includes
#include <core_version.h>                   // Arduino_Esp8266 version information (ARDUINO_ESP8266_RELEASE and ARDUINO_ESP8266_RELEASE_2_3_0)
#include "sonoff.h"                         // Enumeration used in user_config.h
#include "user_config.h"                    // Fixed user configurable options
#include "user_config_override.h"           // Configuration overrides for user_config.h
#include "i18n.h"                           // Language support configured by user_config.h
#include "sonoff_template.h"                // Hardware configuration
#include "sonoff_post.h"                    // Configuration overrides for all previous includes

reloxx13, 2018-03-02:
heya, 

if you are using atom, you could try the latest framework dev version with this:

add this git path in the platformio.ini file, you can use the latest dev build from the arduino framework.

but beware, its alpha/beta status!!!

also, i had to change the cfg_holder once, without the device crashed after few seconds.
```
[env:sonoff]
;platform = espressif8266@1.5.0  ; v2.3.0
;platform = espressif8266@1.6.0  ; v2.4.0
;platform = espressif8266
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
```

tomgregg, 2018-03-05:
Hi!  I should change the title to "Sonoff-Tasmota (2.3.0) does not work with WAN, only with LAN."  I've tried pinging my domain name from multiple access points, some as far as 100 km away, and it always works OK.  Ping times seem reasonable.   However, all three of my ESP8266 modules are becoming less and less reliable over the WAN.  As I said, LAN always works.  One interesting thing is that there is one access point about 10 km away that always works will all my ESP8266's.
Is 2.4.1 ready?  Thank you.

tomgregg, 2018-03-26:
Hi,
I have some more information on this problem.
1) I'm using a WeMos D1 R2 & mini module.
2) Arduino 1.8.5 with ESP8266 version 2.4.0
3) Sonoff version "#define VERSION                0x050C0000   // 5.12.0"
4) I'm trying to get this to work as a server before I attempt an MQTT server.

In the user.config.h file:
1) Set my SSID and PASSWORD
2) Set the port number (#define WEB_PORT) to something in the 7000 range (set my router forwarding accordingly).
3) Disabled MQTT (#define MQTT_USE   0)
4) Enabled a password (WEB_USERNAME         "admin";  WEB_PASSWORD   "password"
Nothing else in the user.config.h file was changed, butI also cleared out the residual flash configuration data with "#define CFG_HOLDER  0x20180324"
I can access my webcams perfectly through my LAN and the WAN.  So, I know I'm doing the right setup for my router. 
Accessing my WeMos module has the problems.  LAN access works perfectly.  WAN access has problems:
1) From the WAN (and LAN too), the module asks for the admin name and password.  When I give it the wrong combination, it asks again (and again and again).  On the WAN, once I enter the correct name and password, it sends back a corrupted web page, and after a while, it simply stops responding.  I went back to the LAN and changed the password.  Once again, from the WAN, it asks for the name and password, and once it gets a valid name and password, to sends back the corrupted page.  (see below).  
After the module gets into a mode where it simply does not respond, I can get it to respond with corrupted web pages only AFTER I reflash with the #define CFG_HOLDER parameter set to wipe out the old configuration data.
One last observation.  I've used several different WAP APs, and they all have the same behavior EXCEPT ONE.  It's a "guest AP" at a local establishment with the IP address 69.206.185.194.  And it's always worked with this IP address for at least 5 or 6 weeks.
Is there any more information I can give you?  Thank you.
[Arduino Screen Shots1.pdf](https://github.com/arendst/Sonoff-Tasmota/files/1849112/Arduino.Screen.Shots1.pdf)

tomgregg, 2018-03-29:
Because of lack of activity, I'm going to close this issue and open a new one with a more appropriate title and include the results from my latest experiments.

closed 2018-03-29 by tomgregg