arendst/Tasmota #2062: Trying to Flash Sonoff Basic

veekay1234, 2018-03-02:
My first attempt at flashing Sonoff.  Windows 10.  Using FTD1232.  Arduino 1.8.5.
Any help is appreciated. 
[image]

 Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, nodemcu, 26 MHz, 40MHz, DOUT, 1M (no SPIFFS), v1.4 Prebuilt, Disabled, None, 115200"

Sketch uses 519612 bytes (50%) of program storage space. Maximum is 1023984 bytes.
Global variables use 51876 bytes (63%) of dynamic memory, leaving 30044 bytes for local variables. Maximum is 81920 bytes.
C:\Users\lenovo-Flex2\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.12/esptool.exe -vv -cd nodemcu -cb 115200 -cp COM4 -ca 0x00000 -cf C:\Users\LENOVO~1\AppData\Local\Temp\arduino_build_868828/sonoff.ino.bin 
esptool v0.4.12 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from  to COM4
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 523760 bytes from C:\Users\LENOVO~1\AppData\Local\Temp\arduino_build_868828/sonoff.ino.bin to flash at 0x00000000
	erasing flash
	size: 07fdf0 address: 000000
	first_sector_index: 0
	total_sector_count: 128
	head_sector_count: 16
	adjusted_sector_count: 112
	erase_size: 070000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting serial port timeouts to 15000 ms
	setting serial port timeouts to 1000 ms
	espcomm_send_command: receiving 2 bytes of data
	writing flash
.............................................................	read 0, requested 1
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
closing bootloader
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

dYna77, 2018-03-02:
Did you set the Sonoff in programming mode by pushing the button (GPIO0 to GND)  while/before powering it up?

jeylites, 2018-03-02:
In addition to @dYna77 post make sure you have the RX and TX in the right ports. Check wiki for more details.  

USB FTDI (RX) -> (TX) Sonoff
USB FTDI (TX) -> (RX) Sonoff

veekay1234, 2018-03-03:
Thanks for your suggestions.  I had followed the correct process.  Fortunately I was able to install the firmware by using my desktop computer instead of laptop.  Also it runs on Windows 7 32bit.  Not sure why one worked and the  other didn't.

closed 2018-03-03 by veekay1234