arendst/Tasmota #2064: rebooting always with new code Version 5.12.0d-2_4_0

upendrawww9, 2018-03-02:
I am getting the following issue
With same credentials works with old release

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_5E2D9F, GroupTopic sonoffs) Version 5.12.0d-2_4_0
00:00:00 WIF: Connecting to AP1 upen-Mifi in mode 11N as sonoff-3487...
00:00:03 MQT: Attempting connection...
08:00:08 MQT: Connected
08:00:08 MQT: tele/sonoff/LWT = Online (retained)
08:00:08 MQT: cmnd/sonoff/POWER = 
08:00:08 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"5.12.0d","FallbackTopic":"DVES_5E2D9F","GroupTopic":"sonoffs"}
08:00:08 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-3487","IPAddress":"192.168.43.33"}
08:00:08 MQT: tele/sonoff/INFO3 = {"RestartReason":"External System"}
08:00:08 WIF: Connected

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_5E2D9F, GroupTopic sonoffs) Version 5.12.0d-2_4_0
00:00:00 WIF: Connecting to AP1 upen-Mifi in mode 11N as sonoff-3487...
00:00:06 WIF: Connected

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9

Jason2866, 2018-03-02:
Which platform (IDE) do you use?
For Arduino be shure that you use stable lwIP version v1.4 (#1940)

upendrawww9, 2018-03-02:
Yes, It worked, I am using Arduino IDE.

meikelmoeller, 2018-03-03:
In my opinion, version 5.12.0 is extremely unstable. Permanent restarts and poor performance. With version 5.11.0 everything runs as usual.
With V 5.12.0 a Sonoff had more than 1200 restarts, another module over 600. 
With 5.11.0 the modules run stable.

johoberm, 2018-03-03:
@meikelmoeller 5.12.0 could be unstable. Generally it works. But you have to watch to use the correct IwiP Version and have a look at free Memory. If using Atom.Io Tasmota takes more care of that...
Using Dev Build 5.12.0b or c with enough free memory works rock solid as 5.11.0.

19 days and not even one restart.

Program Version | 5.12.0
2018-02-10T17:19:53
2_3_0/1.5.3(aec24ac9)
19T21:23:45
Program Size  487kB
Free Memory 21kB

upendrawww9, 2018-03-06:
Dear sir

Greetings from India

Do you have any centers in India.

Currently l work for Reliance Group in 4G domain as solution design lead
Microwave/UBR and wifi domain in plan to build team.
India is fast growing in terms of technology and population.

Do you see any market for IOT.Can you share any exp revenue model with IOT.

I own a small company and planning to start IOT smart home business.Not fun
but planning to have 1 million devices in india.with low marginal.

Request you to share your ideas on this.This will help me

If not wrong whats your revenue model.

Thanks
upendra