arendst/Tasmota #2065: New functionality adding request - attachement MQ sensor Vcc measure MOISURE sens.

roblad, 2018-03-02:
Hi,

Could you review and add attached ino scheches to Tasmota ?

[gassensor.xlsx](https://github.com/arendst/Sonoff-Tasmota/files/1775274/gassensor.xlsx)

[sonoff.zip](https://github.com/arendst/Sonoff-Tasmota/files/1775279/sonoff.zip)

Platform.ini need to have:

[common]
lib_deps_external =
       https://github.com/Apollon77/I2CSoilMoistureSensor

[...]

lib_deps = [...], I2CSoilMoistureSensor
           ${common.lib_deps_external}

arendst, 2018-03-03:
Will look into it...

roblad, 2018-03-10:
Hi,

There is an update finally working version.

[sonoff.zip](https://github.com/arendst/Sonoff-Tasmota/files/1799843/sonoff.zip)

arendst, 2018-03-11:
Looked at it and see some challenges:
- moisture sensor uses to many yields and delays. They need to be removed as there must be a better way to retrieve the information from the sensor.
- voltage is just an analog value to be calculated by your home automation system
- MQ sensor uses code heavy pow and log functions. There must be a better way to do calculations.

Sorry.

roblad, 2018-03-11:
Hi,
>>> moisture sensor 
There is specific of that sensor where delay is main function for that :-) that need even 3-5 sec to get light measure.
See that:
 
https://www.tindie.com/products/miceuz/i2c-soil-moisture-sensor/

Yield must be only one for chirp_readLux() function because of hardware watchdog and in chirp_detect() when reboots sensor. Wait time is needed in 500 milis for waiting a sensor responding in 2 sections of presentation function, additionally there is Wire.setClockStretchLimit(3000); setting needed because of 16mHz clock cycle of Attiny441 and some bug in ESP i2c library. In sensor.getVersion(); function there is as a switch on of that sensor because of wake it up from sleep the time needed is betwen 20-100 ms.

I commented and tested all unnecessary yields ( I will observe what happened when full dark will cause)  and it works, see attachment.
[xsns_20_chirp_mois.zip](https://github.com/arendst/Sonoff-Tasmota/files/1800318/xsns_20_chirp_mois.zip)

>>> voltage is just...

there is showing ADC calculated voltage divider wit the parameters for resistors described, many people asked about it.

>>>MQ sensor uses code heavy pow and log functions...

there is no any other way to get ppm from the sensor is designed for logarithmic scale (I attached links and xlsx file with calculation). The pow function according to slope of the trend line of logarithmic function calculates the ppm values. In my sonoff 4ch there is no any problems to calculate it shows changes very fast.

Would you be so nice and once again consider adding these functionalities ?

Regards
RL

roblad, 2018-06-06:
Hi,

I modyfied code for that sensor 

[xsns_92_chirp_mois.zip](https://github.com/arendst/Sonoff-Tasmota/files/2077856/xsns_92_chirp_mois.zip)

roblad, 2018-06-06:
Hi,

And for MQ sensor - tested and works properly

[xsns_94_MQ_SENSOR.zip](https://github.com/arendst/Sonoff-Tasmota/files/2077860/xsns_94_MQ_SENSOR.zip)

tatunts, 2018-07-19:
Hello @roblad! How are you today?

Sorry to bother you, I am not sure if this is too off topic here, but I was wondering if you could help me in any way.

I've bought this [moisture sensor](https://www.gearbest.com/sensors/pp_1633075.html?wid=1433363) and really want to integrate with my home automation system (Home Assistant) using a cheap ESP8266 D1 Mini. Is there a way to do it, possibly with analog input to see how much moisture the soil has?

I am not a very experienced user and just started using these sketches to flash a few sonoffs and ESPs to connect with HA. Just love using TASMOTA code...

Thanks a lot mate!

roblad, 2018-07-19:
Hi,

There is my source

https://github.com/roblad/Sonoff-Tasmota

tatunts, 2018-07-19:
Thanks! I've downloaded everything on a zip and opened with VSC with PlatformIO, will compile just like I did before. I just don't know how to integrate this voltage ino. So sorry, I had just started learning those things this year hehe

tatunts, 2018-07-20:
Hello @roblad! So sorry to bother you again: I was not able to compile your project... I've just started to tinker with this stuff :)

Would you mind create a standard bin file with no settings, analog input for the sensor and WIFI_MANAGER enabled for a wemos D1 mini? So I can just flash it and plug my moisture sensor for a try...

Really sorry to ask you this, I still have a lot to learn here.

Thanks a lot! 👍 👍 👍

madmicio, 2018-10-12:
Hi!
I'm definitely inexperienced. I bought the sensor.
https://www.tindie.com/products/miceuz/i2c-soil-moisture-sensor/
I did not make any changes to the firmware
I tried to connect it to a basic sonoff. I disabled the seriallog and connected vcc to vcc, gnd to gnd, sda to gpio1 and sck to gpio3.
it does not happen in the.
I tried to connect it to wemos d1 mini. vcc to 5v, gnd to gnd, sda to gpio4 scl gpio5.
once again he does not help anything.
where am I wrong? what is the correct procedure?

madmicio, 2018-10-23:
> [...]
lib_deps = [...], I2CSoilMoistureSensor
${common.lib_deps_external}

can you explain this to me?

ascillato2, 2018-11-06:
Hi, there is someone working on a Tasmota's driver for this sensor?

ascillato2, 2019-01-04:
Hi,

Thanks everyone for sharing their ideas.

Closing this request as it has no activity. Anyway it is added the label REQUEST FEATURE (HOLD OVER) for future reference.

As a workaround for now to read from this sensor is to use the ADC and do the calculations on your home automation software.

If someone wants to make a non-blocking driver for Tasmota, just ask to reopen. Thanks.

Staars, 2019-08-05:
Hi,

if there is still interest here is a possible solution.

[image]

I started with a new driver from scratch and it is non-blocking. We can already read all data, use wake/sleep and change the I2C-address. I still do some tests with auto-sleep and did not do post-processing of the light level data, which are NOT in LUX.
In theory multiple sensors are supported, but I only have one.

ascillato, 2019-08-05:
@Staars 

Cool !

Please, can you make a pull request? Thanks

Staars, 2019-08-06:
Okay, after some clean up  I will make a PR. 
Most of the things left are more or less design decisions (I hope).

ascillato2, 2019-08-06:
Reopening...

Staars, 2019-08-07:
PR is already merged.

I will add something to the wiki in the next few hours. This issue can be closed.

ascillato, 2019-08-07:
Thanks a lot :+1:

closed 2019-08-07 by ascillato2

ptath, 2019-10-12:
> PR is already merged.
> 
> I will add something to the wiki in the next few hours. This issue can be closed.

Hello, can someone please give me any instructions? Cannot find wiki page, trying to connect mq-2 and mq-7.

ascillato, 2019-10-12:
https://github.com/arendst/Sonoff-Tasmota/wiki/Moisture-Sensor-and-Chirp!-Sensor

dmagraf, 2020-11-30:
hi, tell me where to find information about the sensor mq-7

ascillato, 2020-11-30:
The new docs site is at Tasmota.com

dmagraf, 2020-11-30:
> The new docs site is at http://Tasmota.com

Thanks for the link! only there is no information at all about the MQ-7. Maybe I'm just looking for the wrong thing?

ascillato, 2020-11-30:
> there is no information at all about the MQ-7

?????

It is in the list of supported peripherials:
https://tasmota.github.io/docs/Supported-Peripherals/

And it also has its own article with full explanations:
https://tasmota.github.io/docs/Moisture-Sensor-and-Chirp%21-Sensor/

dmagraf, 2020-11-30:
I am very embarrassed, but I did not see a mention of MQ-7 there, so I was misled, thanks I will try to figure out what and how

ascillato, 2020-12-01:
Ok. I found the misunderstanding. The title of this issue is wrong. Here is the request for 2 different sensors. Only chirp that I2C was added to Tasmota. MQ-7 was never implemented. As this issue was closed after the chirp support was added, it appears that also MQ-7 support was added but actually that was not the case.

So, Tasmota supports the Chirp Moisture sensor and that was about this issue and request. I'm changing the title of this issue to reflect the correct request and support added.

The MQ-7 that is a CO sensor and uses an analog value. That value can be read by Tasmota and can be adjusted by commands. Then the threshold levels and alarm messages can be set using Tasmota rules.

Sorry for the misunderstanding.