arendst/Tasmota #2066: Smoothly dimming 3.3/5/12v leds

Joshfindit, 2018-03-02:
(Similar to but not the same as #469)

Apologies if this has been asked before; I've been googling and going through this wiki and I can't find info on this.

- I'd easily pay over $20 for a single Tasmota-powered wifi dimmer device, even if it was only DC-DC
- Willing to wait for shipping
- If possible, I would like to avoid PWM (and definitely avoid low-frequency PWM)
- Using a prototype board, I've had some success with wiring a 3.3V string to 3 GPIO pins and doing rough dimming (off + 3 levels) through limiting the current this way without changing the voltage.
- A relay turns things on or off, but I honestly don't even know the proper term for something that smoothly limits either voltage or current (or whether they exist).

flaotte, 2018-03-04:
For bigger leds PSU (power supply unit) will be bigger than sonoff module itself.

Otherwise you can have powerful PSU and transistors for driving load, and control it from sonoff chip. Smth like voltage regulator with voltage regulator uC and final stage transistors:
http://www.circuitstoday.com/12v-15a-voltage-regulator

But this would be voltage, not current regulation.

p.s. I am not sure if sonoff output is decoupled from mains, you need to check it before attaching anything external.

Joshfindit, 2018-03-04:
For the sake of prototyping, let's limit the request to 3.3V ~20MA 5mm LEDs

@flaotte Excuse my layman's view, but you referenced voltage regulators, and from what I can see those turn variable (or different) voltage in to a single expected voltage (12V in the case you linked to). If that's the case, a voltage regulator does not seem to fit my request.

Joshfindit, 2018-03-15:
@prabbit237 Separately, yes (along with APA102s), however that's way more complexity and cost than is necessary for some simple mood lighting.

Joshfindit, 2018-06-06:
Still interested in this

Joshfindit, 2018-06-06:
**WS2812**
- Still uses PWM (though less noticeable as it's at a higher frequency)
- Needs specific commands to be sent in order to change brightness
- Expensive
- Bulky (comparatively)
- Very difficult to make it look good without hiding it

**'Fairy' lights**
[image]
- Low power even in series
- Pleasant warm-white colour
- Cheap ($0.80/M)
- Can be exposed and still have a nice effect
- A string can be driven using the same circuit designed for a single 5mm LED
- Extremely tolerant, easily replaceable

Ideally, the end result is a small board that has a single variable voltage or variable amperage, and exposes a 'dimmer' value through Tasmota.
For example, imagine a shelf of fairy light jars that dim smoothly and simultaneously in response to a hass scene.

Or plug/play literally any design from sites like http://www.sheknows.com/home-and-gardening/articles/1083360/how-to-decorate-with-fairy-lights

ascillato, 2018-06-06:
Ah, ok

Now I understand. Thanks for share the link.

Tasmota can drive any GPIO as ON, OFF or PWM

If you don't want PWM directly, you will need extra hardware. Have you tried to filter the PWM signal with a R-C filter? That will smooth the signal and will make like if the pwm value is a DC variable voltage source.

```
GPIO ---- RESISTOR ---- Fairy Lights ---- GROUND
                    |
                    |
                CAPACITOR
                    |
                    |
                 GROUND
```

If that is not enough power you will need to drive a MOSFET with this filter circuit

prabbit237, 2018-06-13:
**"Ideally, the end result is a small board that has a single variable voltage or variable amperage"**
Sorry to be the bearer of bad news, but the fairy lights ARE LEDs. So they have a set voltage drop and thus that has to be taken into account. Figuring a 150ohm resistor built in and a 3.3v drop (they're white leds so that's a fairly safe assumption) across the leds, 5v gives just over 11ma (I don't know what the actual resistor and voltage drop are so was just throwing those out as possibles.) Getting down to 3.4 v does drop it down  to a bit under 1ma. So you may not be able to get the dimness you want easily and even with an R-C, for lower PWM rates, you'd still possibly have some ripple in the current to the LEDs. Trying to make it an adjustable constant-current board would increase the cost more.

As to "I don't want PWM", Tasmota can go up to 4khz 'out of the box.' ("PwmFrequency | 100..4000 | Set PWM frequency from minimum 100Hz to maximum 4kHz." Although I can't see how ANYONE could see flicker in that, you can change the code and get up to around 72khz (but probably with some glitching and timing issues due to the uC needing to do other stuff as well.)

ascillato2, 2018-07-21:
@Joshfindit 

Have you managed to solve this?

ascillato2, 2018-08-27:
Closing issue as there is no feedback. Please, ask to reopen if needed. Thanks

closed 2018-08-27 by ascillato2

Joshfindit, 2019-06-04:
To close the loop on this: I'm now using ESP32s as they can go to `312500Hz`! at 8-bit
I'm also using esphome which makes it fairly trivial use logic to control the fairy-light LEDs (I've used homeassistant, MQTT, rotary encoders, potentiometers, and capacitive touch so far)

One drawback: these LEDs are actually happier being powered with 3.6-4.5V, NOT the 3.3 that most boards output. I've yet to experiment with voltage shifters, but I feel like they would work well as long as the part can handle the PWM speed. Looking at "logic shifters" for this as they support high frequency by design.