arendst/Tasmota #2068: Wemos D1 Mini Pinout Problem

Pilzbauer90, 2018-03-02:
Hello,
I have my first Wemos D1 Mini and managed to install Tasmota on it.
Then I tried to run a relay with it but every time I put a pin to relay1 the pin get power but I cant toggle it.
I run the chip with 5V from an external power supply.
Is there anyone who can help me?

Jason2866, 2018-03-03:
Do you have connected the relay direct to the pin without a driver transistor?
If yes that do not work. The relay needs to much current.

Pilzbauer90, 2018-03-03:
I have a relay board like this
https://images-na.ssl-images-amazon.com/images/I/61xUjEcTHPL._SX342_.jpg

The problem is that the pin is always high and I can’t toggle it to off.

johoberm, 2018-03-03:
So, let´s do one step by another. Disconnect the relay. Just power it with the 5 Volt power supply.
Now connect GND to Input. Does something happen? Now connect +5 Volt to Input. Does something happen? One answer must be yes! Does the relay Switch with GND or with 5V?

johoberm, 2018-03-03:
Next step. Connect GND Modul with GND relay. Connect D4 (GPIO 2) to relay INPUT.
D4 has a pullup resistor already on the circuit board, and a LED on the ESP. You see whats goes on...
https://wiki.wemos.cc/products:d1:d1_mini_pro
On the web page of Tasmota Firmware select module "Generic"
Select GPIO2 "Relay1" and save it. Thats all. If ON and OFF isnt correct select for GPIO2 "Relay1i"

Pilzbauer90, 2018-03-03:
Thank you. 
Now I can toggle the build in LED but the relay stays on. Here a picture of the setup. 
https://www.dropbox.com/s/4a17014e7dotrd1/Foto%2003.03.18%2C%2020%2052%2040.jpg?dl=0
I don’t know what’s wrong. Maybe the relay needs an additional resistor or anything like this.

Jason2866, 2018-03-03:
Do you have a voltmeter? If yes measure the input of the relay, when on and off.
Have you tried the relay stand alone as i described above?

KPK75, 2018-03-03:
It looks just like one I have. Mine does not work work with 3v3 signals on IN. You need to put 5v to IN to turn it off.

Pilzbauer90, 2018-03-03:
When I remove the cable from INN then the relay turns off

Jason2866, 2018-03-03:
Disconnect the wemos. Jumper you power supply for the breadboard one side 3.3V and the other with 5V
Power the relay with 5V. Put the 3.3V to the Input? Does it switch with 3.3V?

Pilzbauer90, 2018-03-03:
I just figured out that my relay a low level trigger. 
I tested it with another relay I borrowed from a friend and it worked as expected. Thank you all for your help. I think I can figure it out myself how it will work with this one. I thought it was a problem with the board or the program.

closed 2018-03-03 by Pilzbauer90