arendst/Tasmota #2069: Boiler Controls

dawsmac, 2018-03-02:
Hi All, 
 
Not sure if this is possible or I am not understand something.

OK, I have a Sonoff TH setup to fire up the Boiler when needed, this work fine.  I also have a PIR Sensor setup to pick up motion this also works fine ;). now the problem comes when I either shut down the MQTT services to do an update (Hassio) it seems like the PIR takes over and starts to fire up the boiler...  for Safty reason I want when the wireless is down or MQTT server it not advisable the Sonoff will switch off is this possible? 

Thanks 

Rich

mkh595, 2018-03-02:
it is same behaviour like described here #1614

You can try to assign PIR to Switch2, which will not control Relay1 if MQTT is not connected.

dawsmac, 2018-03-03:
Hey Mkh595,  

I should have said I have the PIR setup with Switch2.. is it something to do with when not connected to Wifi it fails back to defaults.

Cheers

Rich