arendst/Tasmota #2071: MQTT response to command not following fulltopic template

Dawnes, 2018-03-03:
Sonoff Basic tested with 5.11.1 and 5.12 versions, same behaviour.

Fulltopic is set to: home/%topic%/%prefix%
where topic = ikea and prefix = default values.

Local control with the button gives a MQQT response in the correct topic:
home/ikea/stat/POWER ON

However, a response to an MQTT command message goes to the wrong topic:
home/ikea/cmnd/POWER on
stat/ikea/RESULT {"POWER":"ON"}
stat/ikea/POWER ON

This doesn't seem correct.

arendst, 2018-03-03:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

Dawnes, 2018-03-03:
Hardware is a Sonoff Basic. Command output for status 0 is below. There is no crash so no stack dump and debug messages require a serial connection which is hard because the device is built in.

I thought my report was pretty much complete, what info are you missing in order to look into this?

`19:20:55 CMD: status 0
19:20:55 MQT: stat/ikea/STATUS = {"Status":{"Module":1,"FriendlyName":"ikea","Topic":"ikea","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
19:20:55 MQT: stat/ikea/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","Uptime":"0T17:52:16","Sleep":0,"BootCount":359,"SaveCount":1156,"SaveAddress":"F5000"}}
19:20:55 MQT: stat/ikea/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:54:51","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:20:55 MQT: stat/ikea/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"NoWire","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00000029"}}
19:20:55 MQT: stat/ikea/STATUS4 = {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
19:20:55 MQT: stat/ikea/STATUS5 = {"StatusNET":{"Hostname":"ikea-2056","IPAddress":"192.168.5.63","Gateway":"192.168.5.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.5.250","Mac":"EC:FA:BC:00:88:08","Webserver":2,"WifiConfig":2}}
19:20:55 MQT: stat/ikea/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.5.66","MqttPort":1883,"MqttClientMask":"ikea","MqttClient":"ikea","MqttUser":"ikea","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:20:55 MQT: stat/ikea/STATUS7 = {"StatusTIM":{"UTC":"Sat Mar 03 18:20:55 2018","Local":"Sat Mar 03 19:20:55 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
19:20:55 MQT: stat/ikea/STATUS10 = {"StatusSNS":{"Time":"2018-03-03T19:20:55"}}
19:20:55 MQT: stat/ikea/STATUS11 = {"StatusSTS":{"Time":"2018-03-03T19:20:55","Uptime":"0T17:52:16","Vcc":3.222,"POWER":"ON","Wifi":{"AP":1,"SSId":"NoWire","RSSI":66,"APMac":"E0:28:6D:1A:11:97"}}}`

arendst, 2018-03-03:
Make clientid different from topic and your problem is solved.

You're one of many hitting this feature which has been dicussed here at least in ten issues. Just leave clientid as original defined (unique based on device mac address).

Dawnes, 2018-03-03:
Je hebt helemaal gelijk. This workaround solved the issue for me. Thanks.

closed 2018-03-09 by arendst