arendst/Tasmota #2072: How to enter serial commands?

Zixim, 2018-03-03:
Hello,
when I connect to the serial console, Putty is displaying the serial log.
This leads me to believe that there's a good serial connection and that the sonoff (basic) is alive & well.

How would I go about sending a command over this serial connection ?
For instance, I type AP in the console, but nothing appears on screen... hit enter...nothing... hit any key....nothing.

What am I missing ?

arendst, 2018-03-03:
Make sure seriallog is at least set to 2

Zixim, 2018-03-03:
but, I **_am_** seeing the seriallog in Putty's window.

The thing is, I **can't** use the serial console to issue commands.

arendst, 2018-03-03:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``

arendst, 2018-03-03:
Use another tool like Termite

Zixim, 2018-03-03:
it is a Sonoff Basic, as mentioned in original post.
I try to type ``status 0`` in the serial console, but nothing appears, only the Wifi connection status scrolls in window.

Zixim, 2018-03-03:
Thanks on the Termite tip. This works.... this must be the 1st time that Putty has let me down.

closed 2018-03-03 by Zixim

yearski, 2019-01-24:
PuTTY can work (but not "ideal"). There are two issues:
1. You need local echo to see that typed characters are actually doing something: Configuration > Terminal > Local echo > Force on
2. You need a LF line feed character to end the command. CTRL-J sends this.

There is currently no option in PuTTY to automatically send the LF with the Enter key (this is an option in Termite hence why it just works). I've sent a feature request to the PuTTY team for this. For the little use I have, just using CTRL-J is acceptable

Race666, 2020-12-21:
May I ask developers: What was the design decision to use LF as "enter" and not CR like all others do?

lalo-uy, 2020-12-21:
Define "all others"?
LF is the end of line standar on Linux and Java.
Also on Arduino terminal ( since it is on Java)

Race666, 2020-12-21:
Sure? https://unix.stackexchange.com/questions/349214/enter-key-is-different-from-carriage-return-cr
I think you mean the new line in a file.

Race666, 2020-12-21:
This is defined in the ASCII Table https://www.rapidtables.com/code/text/ascii/ascii-enter.html CR = Enter

eadmaster, 2024-05-28:
for me the problem was the serial speed set uncorrectly (must be 115200)