arendst/Tasmota #2075: New functionality adding request: invert led state

Vikingfr, 2018-03-03:
Hi,
I'm using sonoff+tasmota to control my heater using his control wire (we call it 'fil pilote' in France). If relay is off, heater is working, if relay is on, heater stop.
Could you add option to invert led state, so LED is on if relay is off. This way, I know heater is ON when LED is visible

(sorry if bad english, I'm french)

Jason2866, 2018-03-03:
It is already implemented. Use generic module. Here you can define led inverted -> Led1i

Jason2866, 2018-03-04:
It is there as a software option! you have to change it in the web console of the module as i described.
Read the wiki...

geek65535, 2018-03-10:
@Jason2866 Rather than being snippy at someone who doesn't speak English as his first language, perhaps you could be a little more helpful and point to the part of the wiki that explains how to invert the output of the LED. I'd do it, but I can't find the information myself.

Jason2866, 2018-03-10:
Sorry, if i was unpolite. English isnt my mother language too....

If Tasmota is running on the module you can select which type of module to use.
Dont select here the dual. Select modul generic.
Take a look in the source code template.h. Here you find the definitions for the GPIO functions for the modules. Use the definition Sonoff dual as start for your settings in the Web Console for the settings
Module Generic. If switching works use instead as Example for Led1 Led1i. This will invert the Led function

OneHoopyFrood, 2020-09-22:
If anyone else finds this like I did, and all you want to do is flip the LED on state, the answer is templating.

Steps to configure in this manner:
1. Go to configuration -> Configure Template
2. Give it a name (defaults to Generic)
3. Set Based On to Sonoff Basic (this is what I'm using, if you're not then you'll need to adapt accordingly)
4. Set GPIO13 to Led1 (defaults to Led1i)
5. Save -> Return to main menu after restart
6. Go to configuration -> Configure Module
7. Set module type to your template name from step 2
8. Save

The LED should now be on when the relay is off, and vice versa.