arendst/Tasmota #2076: Sonoff POW and S31, wifi led not working with switch off.

mattlward, 2018-03-03:
Make sure these boxes are checked before submitting your issue - Thank you!

The hardware is working fine on all 3 units, but the blue light is only lit with the devices relay switch is turned on.  The POW was OTA'd with 5.12 and the S31's were serial programed with sonoff.bin via esptool.py

- [ ] What hardware you are using
Sonoff POW and 2 S31's
- [ ] Provide the output of command ``status 0``
`
23:29:44 CMD: status 0
23:29:44 RSL: stat/sonoff/STATUS = {"Status":{"Module":41,"FriendlyName":"S31-1","Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
23:29:44 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T00:29:31","Sleep":0,"BootCount":49,"SaveCount":125,"SaveAddress":"FB000"}}
23:29:44 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:54:51","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
23:29:44 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"dd-wrt_vap","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00200009"}}
23:29:44 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
23:29:44 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0936","IPAddress":"192.168.1.102","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.4","Mac":"5C:CF:7F:7D:23:A8","Webserver":2,"WifiConfig":3}}
23:29:44 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"domus1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7D23A8","MqttUser":"DVES_USER","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:29:44 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Mar 03 22:29:44 2018","Local":"Sat Mar 03 23:29:44 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
23:29:44 RSL: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
23:29:44 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-03-03T23:29:44","ENERGY":{"Total":0.003,"Yesterday":0.000,"Today":0.003,"Power":0,"Factor":0.00,"Voltage":125,"Current":0.000}}}
23:29:44 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-03-03T23:29:44","Uptime":"0T00:29:31","Vcc":3.176,"POWER":"ON","Wifi":{"AP":1,"SSId":"dd-wrt_vap","RSSI":100,"APMac":"BA:A3:86:50:6B:00"}}}
`
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

arendst, 2018-03-04:
As designed. It only shows power on/off. Look at command ledstate.

roguestreak, 2018-03-04:
I can't remember what I had to do on my S22, but did find a way to get the wifi LED working; may have been setting it up as a sonoff touch?
That said, I'm not sure if it's actually working as a wifi indicator or just power supplied?

Theo, as far as I could tell from the commands and experiments there are only options for the one LED, not two like in the POW / S22 / (and guessing S31?)?  Is there a way of targeting each LED for a specific task?  On the stock firmware the blue LED indicated the wifi connection state, and the red indicated the relay status; seems the red is the only one affected by ledstate?

mattlward, 2018-03-05:
Yes, the S31 has 2 leds.  In stock config the blue one is a wifi indicator and the red one is the relay state indicator.