arendst/Tasmota #2078: Can't access web gui from smartphone

Cysign, 2018-03-04:
Make sure these boxes are checked before submitting your issue - Thank you!

- [x] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

Whilst I'm able to access the IP of my Tasmota Sonoff Basic with my tablet (Medion Lifetab X10302, Android 7.0) properly, my Smartphone (Huawei P10, Android 7.0) cannot access it. I'm always getting a timeout (both, using Chrome and Firefox). 
Using Fing I can scan my network and see the device on it's IP as 'Espressiv' (I also can find my tablet - so I'm sure there's nothing wrong with IP addresses or subnetmask).

Cysign, 2018-03-04:
Sonoff Basic is running Tasmota 5.12.0

wayne47git, 2018-03-23:
If you compiled this yourself, please try downloading the pre-built one. Every version I build has this problem but the ones I download work fine.

Cysign, 2018-03-23:
Hmmm....what's the problem with the fresh compiled one using ArduinoIDE?
I have no clue how to flash the precompiled file on my module.

Cysign, 2018-03-28:
I startet the Firmware Update from the gui pointing to http://domus1:80/api/arduino/sonoff.ino.bin - so now the precompiled firmware (even if it's the same as before?) should be installed. How could I prove if it was updated properly?

Still can't access the GUI using my smartphone.