arendst/Tasmota #2081: Hi,

cyberelectronics, 2018-03-05:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

cyberelectronics, 2018-03-05:
Hi,
First I want to say Thank you! for this great Tasmota firmware!
I want to connect an equipment with serial interface (RS232/TTL) to the ESP8266 TX/RX port.
Is it possible to send/receive serial text messages on the ESP8266 TX/RX port, thru MQTT messages?
One of my friends working on Google Assistant SDK implementation on RPi3.
This is what we testing now https://youtu.be/X9inuvuyzHQ    
Thanks!

ionciubotaru, 2018-03-10:
Some suggestions:
- it's better to use `software serial` library https://www.arduino.cc/en/Reference/SoftwareSerial just to keep hardware serial for debugging and uploads.
- consider using `easy transfer` library  https://tech.scargill.net/easytransfer-for-esp8266/, because data transfer result is ready to use inside code (a json string is harder to convert into normal variables)

cyberelectronics, 2018-03-10:
Thanks for your suggestions ionciubotaru!
No need for debugging :) and the uploads will be OTA. 
It's easier to use the RX/TX Header on the Sonoff Basic devices than wiring unconnected pins.
The Sonoff Basic can be a cheap All in One solution to turn On/Off a device + MQTT to Serial conversion + IR command recv/transmit for other equipments like TVs, Receivers, Pellet Stove :) )

ionciubotaru, 2018-03-10:
That's true, for your case a sonoff basic is enough, but using esp8266 as a serial to mqtt transceiver will be useful for other projects too, that's why I suggest to implement enhanced features.

arendst, 2018-03-20:
See 5.12.0h

cyberelectronics, 2018-03-20:
Great!
Thank you Arendst!

closed 2018-06-02 by ascillato