arendst/Tasmota #2083: Sonoff SV Issues When Losing Power to Sonoff

BluePalasky, 2018-03-05:
Hi Guys

I have a Sonoff SV that I'm attempting to use to control my AC unit.

I have the latest tasmota firmware and I'm communicating with Home Assistant via MQQT.

I have flashed 10 sonoff basics and they are working perfectly, but my SV is losing connection to the wifi when it loses power (I could set it up with permanent power to the device but on a power outage this wouldn't be convenient!).

I have removed the two resistors from the PCB to enable external powering of the device.

Configuration:
Module: Sonoff SV
GPIO14: Switch 2

Console:
pulsetime 3
switchmode1 0
switchmode2 2
switchtopic2 2

Unfortunately I have had a large number of connectivity issues with this device including:

Typically when I configure the MQQT settings it will become quite unresponsive and lose connection a lot - even though accessing and navigating the main page before changing the settings worked seamlessly.  

Persevering with the above issue over a large number of hours I was able to configure the device to work exactly as I wanted - integrated with Home Assistant.  But once I remove the power source and re-connect, my device gets lost.  I either have to re-flash the device or sometimes I can toggle the switch 4 times and that brings it back to life.

Is there a chance my Sonoff SV is faulty or is this a firmware issue?  I took a video of the problem here: https://youtu.be/YEyg4I97yoA 

Thanks for your help in advance guys - love your work!

Cheers

awaldeck, 2018-03-06:
I have seen my SV do exactly the same thing, same settings. If I leave it on for a long time eventually it comes back on its own to the wifi

BluePalasky, 2018-03-07:
Hi awaldeck, that's really interesting - thanks for your response.

I wonder if it's the firmware or the device that's causing this?  I'm not sure if there's another firmware I could try that could make use of the GPIO pins to monitor the state of the equipment.

I reckon I'll just order another SV to eliminate the hardware as the problem.

Cheers

BluePalasky, 2018-03-07:
Hi Guys,

Problem solved!

The issue I believe was power/ voltage related.  Maybe a small voltage drop from my USB 5V connection via jumper wire to the external power supply (which is listed as 5-24V).

I've hooked it up with a 12V supply and now everything works seamlessly and the sonoff reconnects to the wifi immediately after losing power.  

***next thing to figure out is to how to stop the relay firing upon supply resumption.***

Cheers!

closed 2018-03-09 by arendst

BluePalasky, 2018-03-14:
Hi Guys,

Problem solved for 1 week only.  Came home today and my AC unit was togging on and off every 30 seconds.  

Checked the console and its having connection issues again.  Definitely not the power supply as I'm running 12V now.

I have tried:

- restarting Home Assistant

- restarting the sonoff

- restoring tasmota to default settings

- reconfiguring to my required settings

It appears to be firing the relay upon every connection loss but I'd prefer if this didn't happen.  I tried poweronstate 0 but that hasn't stopped the on/off issue every 20 seconds!

Here's an extract from my console - any advice would be appreciated!:

000:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_05307C, GroupTopic sonoffs) Version 5.12.0
00:00:00 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
00:00:00 RSL: stat/sonoff/POWER = OFF
00:00:00 WIF: Connecting to AP1 iiNet3DFA2F in mode 11N as sonoff-4220...
00:00:07 WIF: Connected
00:00:07 DNS: Initialized
00:00:07 HTP: Web server active on sonoff-4220.local with IP address 10.1.1.234
13:27:40 MQT: Attempting connection...
13:27:40 MQT: Connected
13:27:40 MQT: tele/sonoff/LWT = Online (retained)
13:27:40 MQT: cmnd/sonoff/POWER = 
13:27:40 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff SV","Version":"5.12.0","FallbackTopic":"DVES_05307C","GroupTopic":"sonoffs"}
13:27:40 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-4220","IPAddress":"10.1.1.234"}
13:27:40 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
13:27:40 MQT: homeassistant/switch/sonoff/config =  (retained)
13:27:40 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
13:27:40 MQT: stat/sonoff/POWER = ON
13:27:40 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
13:27:40 MQT: stat/sonoff/POWER = ON
13:27:40 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
13:27:40 MQT: stat/sonoff/POWER = OFF
13:27:41 MQT: cmnd/2/POWER2 = OFF
13:27:41 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
13:27:41 MQT: stat/sonoff/POWER = OFF
13:27:48 MQT: tele/sonoff/STATE = {"Time":"2018-03-14T13:27:48","Uptime":"0T00:00:17","Vcc":3.471,"POWER":"OFF","Wifi":{"AP":1,"SSId":"iiNet3DFA2F","RSSI":82,"APMac":"C4:EA:1D:3D:FA:2F"}}
13:27:48 MQT: tele/sonoff/SENSOR = {"Time":"2018-03-14T13:27:48","Switch2":"OFF"}
13:28:03 MQT: Attempting connection...
13:28:03 MQT: Connected
13:28:03 MQT: tele/sonoff/LWT = Online (retained)
13:28:03 MQT: cmnd/sonoff/POWER = 
13:28:03 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
13:28:03 MQT: stat/sonoff/POWER = ON
13:28:03 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
13:28:03 MQT: stat/sonoff/POWER = ON
13:28:03 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
13:28:03 MQT: stat/sonoff/POWER = OFF
13:28:03 MQT: cmnd/2/POWER2 = ON
13:28:40 MQT: Attempting connection...
13:28:40 MQT: Connected
13:28:40 MQT: tele/sonoff/LWT = Online (retained)
13:28:40 MQT: cmnd/sonoff/POWER = 
13:29:04 MQT: Attempting connection...
13:29:04 MQT: Connected
13:29:04 MQT: tele/sonoff/LWT = Online (retained)
13:29:04 MQT: cmnd/sonoff/POWER = 
13:29:06 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
13:29:06 MQT: stat/sonoff/POWER = ON
13:29:06 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
13:29:06 MQT: stat/sonoff/POWER = ON
13:29:06 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
[... 184 lines cut ...]
13:39:47 MQT: Connected
13:39:47 MQT: tele/sonoff/LWT = Online (retained)
13:39:47 MQT: cmnd/sonoff/POWER = 
13:39:49 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
13:39:49 MQT: stat/sonoff/POWER = ON
13:39:49 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
13:39:49 MQT: stat/sonoff/POWER = ON
13:39:49 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
13:39:49 MQT: stat/sonoff/POWER = OFF
13:39:50 MQT: cmnd/2/POWER2 = OFF
13:40:14 MQT: Attempting connection...
13:40:14 MQT: Connected
13:40:14 MQT: tele/sonoff/LWT = Online (retained)
13:40:14 MQT: cmnd/sonoff/POWER = 
13:40:36 MQT: Attempting connection...
13:40:36 MQT: Connected
13:40:36 MQT: tele/sonoff/LWT = Online (retained)
13:40:36 MQT: cmnd/sonoff/POWER = 
13:40:36 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
13:40:36 MQT: stat/sonoff/POWER = ON
13:40:36 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
13:40:36 MQT: stat/sonoff/POWER = ON
13:40:36 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
13:40:36 MQT: stat/sonoff/POWER = OFF
13:40:36 MQT: cmnd/2/POWER2 = ON

awaldeck, 2018-03-14:
The problem is when the MQTT server connection drops, tasmota uses the configured GPIO's in your case as GPIO14 Switch 2 as a switch to trigger the relay, even tho it is not even configured to do so as you have switchmode2 2 and switchtopic2 2 , I experience this with all my sonoffs...
It seems you must never lose connectivity. I don't know if this is by design or if this is a bug in the firmware as no one has responded to my similar issues I have logged and with the Search I cannot get any concrete answers either.

You do however have a connection problem, it might be weak signal or a network issue.

uspass, 2018-09-01:
For connection drop set static IP (not DHCP):
IPAddress1 WhateverLocalIPAddressYouWant
IPAddress2 WhateverGatewayYouMayHaveInYourLocalNetwork

For not triggering relay set:
PowerOnState 0

Read the doc:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands