arendst/Tasmota #2084: MQT: Connect failed to 192.168.178.86:1883, rc -2. Retry in 10 sec

Extinguo, 2018-03-05:
Hey guys,
I´m using two Sonoff Basics. One is in the living room and one is in my bedroom.  All of a sudden the one in  my bedroom stopped working properly. It doesn´t respond to MQTT Commands so I checked the console and got this error: 
```
15:24:51 MQT: Attempting connection...
15:24:57 MQT: Connect failed to 192.168.178.86:1883, rc -2. Retry in 10 sec
```
Looked up the error code:
`-2: MQTT_CONNECT_FAILED - the network connection failed`

this is weird, because I didn´t change anything. 
It´s also hooked up to the wall switch which also doesn´t respond as it used to (sometimes it responds delayed or not at all). This started yesterday. 
Any ideas?

completelyClueless, 2018-03-05:
hi,
I had the same problem a week ago - it was my firewall on the router because I run the Sonoffs in a separate network at home (no access to internal LAN)
KR,
cc

Extinguo, 2018-03-06:
Hey, thanks for you reply. 
What did you change? 

I´ve got a FritzBox 7390 configured as a repeater. This is the one that the not working sonoff connects to. 

Everything is within one network.

stepanovalex, 2018-04-20:
Same problem: https://github.com/arendst/Sonoff-Tasmota/issues/958#issuecomment-383030194