arendst/Tasmota #2088: Configuration problems

ThomasBj, 2018-03-05:
Hi
I have some Sonoff Basic that have been flashed with the lates sonoff.bin firmware version 5.12.0.
However, after pressing the button 4 times and connecting to the accesspoint i can not get the webserver to respond so that i can enter my wifi credentials.
What am i doing wrong?

willyk0, 2018-03-06:
I did this lots of times last week without problems, also with 5.12:

- Pressing 4 times fast (clickclickclickclick), led should be blinking twice per second aprox. Is your sonoff doing this?
- I open the wifi list and refresh until I see a SSID named "sonoff-XXXX", and connect to it. If I can't see this SSID, I disable and reenable wifi on my computer, and if it fails, I plug off the Sonoff and plug it on again, and again I click 4 times. I repeat this until I see the SSID.
- Connect to that SSID. Automatically, a web should pop up asking for wifi credentials.

Which step are you at?

ThomasBj, 2018-03-06:
Thanks for your reply.
Yes, I have connected to the sonoff-XXXX accesspoint but no webpage opens automatically and if i with Chrome or IE go to 192.168.4.1 nothing happens (except for odd occations when the configuration page is shown but later stops responding so my credentials are never stored).