arendst/Tasmota #2089: Random or configurable wait time for WIFI_[RETRY|WAIT] to avoid connection problems

willyk0, 2018-03-05:
I have about 15 Sonoff Basics at home (Tasmota up to date) and I've experienced connection problems with power outages (I've caused lots of them this days manipulating my electrical installation). When the power comes back, every sonoff tries to connect to my wifi at exactly the same time... and routers usually don't behave very well with this and many of them fail to connect (in fact, if they have the "wifi manager" mode configured with WifiConfig 2, they fall in AP mode until they restart themselves). So, I configured WIFI_WAIT (WifiConfig 5) in order to retry the same AP later. I've seen in code that the wait time is a fixed value:

#define WIFI_RETRY_SEC    30  

So, what happens? After wait time, all sonoffs which failed are trying to reconnect to wifi... exactly at the same time again, so many of them fail again... and so, until everyone is configured a bunch of minutes later.

I thought perhaps it would be a good idea to change the wait time with a random value between 15 and 45 secs, or making it a configurable variable instead of a fixed hardcoded constant so I could configure my sonoffs with different values avoiding this problem.

Or maybe I should spend money in a better Access Point :) (currently TP-LINK Archer C9)

jeylites, 2018-03-06:
I’m in the same situation. Need clarification on solve this issue

aferraz, 2018-03-06:
This is a good idea, maybe even MQTT configurable so you can differentiate retry time.

CommodoreWhite, 2018-03-06:
Easiest to base the delay on the unique device id

jeylites, 2018-03-06:
Where is the delay function at?

titomontes, 2018-04-16:
Hello,

How was this issue solved? I´m very interested, because I have the exact same problem. PLease don´t tell me I have to reflash all my sonoffs (20+)...

ascillato, 2018-06-02:
In the file _support.ino_ there is the key `#define WIFI_RETRY_OFFSET_SEC  20   // seconds`

closed 2018-06-02 by ascillato

khriss75, 2018-12-19:
Sorry for this other message, but I don't understand the solution. I have the same problem with more than 20 devices. Can you confirm that the solution is to re-flash all devices with  `#define WIFI_RETRY_OFFSET_SEC 20 // seconds` changing "20" with other different numbers for each devices? No possible via MQTT (like mqttretry...)? I'm on 6.3 firmware.

ascillato, 2018-12-19:
Hi @khriss75

This issue is too old. A lot of things has changed since then. 

Please, if you have a similar problem, update first to latest and try again. Thanks.