arendst/Tasmota #2094: getting state of sonoff

NiekN, 2018-03-06:
I have a PI3
and im using Domoticz to control the sonoff

i installed mosquitto client, so i can switch it on off by
http://myIP/cm?cmnd=POWER%20ON
http://myIP/cm?cmnd=POWER%20OFF

but how can i check the status , do i have to make a time script to check ?

RaymondMouthaan, 2018-03-07:
Hai Niek,

Please read the [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands) its well explained how you can the status of your device. There are basically three ways to give commands, web, mqtt and via serial.

Please take note of the TelePeriod command, this can send the status to mqtt at a period of time. 

btw the above commands you mention are web commands aka http request and not mqtt commands. 

I am not a Domoticz user but what you can do is, to make domoticz subscript to the topic on which your Tasmota device publish its message too. Set the TelePeriod for a given time (for example 30 sec). The device will publish every 30 sec its status to mqtt and domosticz "listens" to it and you can display the status or do anything else with it.

NiekN, 2018-03-08:
Hi RaymondMouthaan 

im a newbee,  an i hoped it was well descripted in the wiki for me , but you need a certain level to inderstand en get it working. so i hoped there were some examples i to set it up in domoticz , otherwise it is more try en error for me