arendst/Tasmota #2095: Sonoff RF Bridge: variable RF Codes

timota, 2018-03-07:
What hardware you are using:

>  - Sonoff RF Bridge.
>  - remote control RSL366T based on SC5262S chip

Provide the output of command ``status 0``

> 01:48:01 RSL: STATUS = {"Status":{"Module":25,"FriendlyName":"Sonoff","Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
> 01:48:01 RSL: STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T00:13:55","StartupUTC":"2018-03-07T00:34:06","Sleep":0,"BootCount":13,"SaveCount":42,"SaveAddress":"F4000"}}
> 01:48:01 RSL: STATUS2 = {"StatusFWR":{"Version":"5.12.0d","BuildDateTime":"2018-02-28T19:46:47","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
> 01:48:01 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"","SSId2":"","TelePeriod":300,"SetOption":"00000001"}}
> 01:48:01 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":496,"Free":504,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
> 01:48:01 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-7478","IPAddress":"192.168.0.83","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:82:5D:36","Webserver":2,"WifiConfig":3}}
> 01:48:01 RSL: STATUS7 = {"StatusTIM":{"UTC":"Wed Mar 07 00:48:01 2018","Local":"Wed Mar 07 01:48:01 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
> 01:48:01 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-03-07T01:48:01"}}
> 01:48:01 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-03-07T01:48:01","Uptime":"0T00:13:55","Vcc":3.504,"Wifi":{"AP":1,"SSId":"","RSSI":68,"APMac":""}}}

Issue: 
 when im pressing Button1 on my remote several times i got strange results:
 

> 01:41:24 RSL: RESULT = {"RfReceived":{"Sync":13450,"Low":430,"High":1290,"Data":"155457","RfKey":"None"}}
> 01:41:27 RSL: RESULT = {"RfReceived":{"Sync":13450,"Low":430,"High":1290,"Data":"155457","RfKey":"None"}}
> 01:41:30 RSL: RESULT = {"RfReceived":{"Sync":13440,"Low":440,"High":1290,"Data":"155457","RfKey":"None"}}
> 01:41:41 RSL: RESULT = {"RfReceived":{"Sync":13470,"Low":430,"High":1300,"Data":"155457","RfKey":"None"}}
> 01:41:46 RSL: RESULT = {"RfReceived":{"Sync":13460,"Low":430,"High":1300,"Data":"155455","RfKey":"None"}}
> 01:41:49 RSL: RESULT = {"RfReceived":{"Sync":13460,"Low":430,"High":1290,"Data":"155457","RfKey":"None"}}
> 01:41:55 RSL: RESULT = {"RfReceived":{"Sync":13450,"Low":430,"High":1290,"Data":"155457","RfKey":"None"}}
> 01:41:59 RSL: RESULT = {"RfReceived":{"Sync":13450,"Low":430,"High":1290,"Data":"155457","RfKey":"None"}}
> 01:42:23 RSL: RESULT = {"RfReceived":{"Sync":13450,"Low":440,"High":1290,"Data":"155457","RfKey":"None"}}
> 01:42:28 RSL: RESULT = {"RfReceived":{"Sync":13460,"Low":430,"High":1290,"Data":"155557","RfKey":"None"}}
> 01:42:58 RSL: RESULT = {"RfReceived":{"Sync":13460,"Low":430,"High":1290,"Data":"155457","RfKey":"None"}}
> 01:43:00 RSL: RESULT = {"RfReceived":{"Sync":13490,"Low":420,"High":1300,"Data":"155455","RfKey":"None"}}
> 01:43:00 RSL: RESULT = {"RfReceived":{"Sync":13440,"Low":440,"High":1290,"Data":"155557","RfKey":"None"}}
> 01:43:09 RSL: RESULT = {"RfReceived":{"Sync":13450,"Low":430,"High":1290,"Data":"155557","RfKey":"None"}}
> 01:43:13 RSL: RESULT = {"RfReceived":{"Sync":13440,"Low":430,"High":1290,"Data":"155557","RfKey":"None"}}
> 01:51:13 RSL: RESULT = {"RfReceived":{"Sync":13520,"Low":430,"High":1300,"Data":"151557","RfKey":"None"}}
> 01:51:16 RSL: RESULT = {"RfReceived":{"Sync":13510,"Low":430,"High":1300,"Data":"151557","RfKey":"None"}}
> 01:51:18 RSL: RESULT = {"RfReceived":{"Sync":13520,"Low":430,"High":1290,"Data":"155557","RfKey":"None"}}
> 01:51:23 RSL: RESULT = {"RfReceived":{"Sync":13530,"Low":430,"High":1300,"Data":"155557","RfKey":"None"}}
> 01:51:25 RSL: RESULT = {"RfReceived":{"Sync":13510,"Low":430,"High":1290,"Data":"155557","RfKey":"None"}}
> 01:51:27 RSL: RESULT = {"RfReceived":{"Sync":13580,"Low":420,"High":1300,"Data":"151557","RfKey":"None"}}
> 01:51:29 RSL: RESULT = {"RfReceived":{"Sync":13530,"Low":430,"High":1290,"Data":"155557","RfKey":"None"}}

as all RfSync, RfLow, RfHigh and last RfCode parameters are different each time i cant use Learn functionality, even cant enter manual them via "RfKey1 4"
Please help how can this issue be addressed and resolved. Is any options how i can catch/learn these codes ?

wongnam, 2018-03-07:
use RfKey1 2 to learn the code from remote.

johoberm, 2018-03-07:
The used code of your remote is probably not compatible with the RF-Bridge

timota, 2018-03-07:
@wongnam yes, i tried to use "RfKey1 2" - when i started the command - Bridge came to Learn mode, but when im pressing the button on my remote - nothing happened and due to timeout Bridge didnt learn rfcode. When im doing the same with remote that came with Bridge - code has been successfully learned.  

@johoberm - yes this is Power Outlet Switches with DIP. Can you please point me on the right direction how i can calculate codes myself. 

BTW: i can sniff/send codes via Arduino RCSwitch library.

Jason2866, 2018-03-07:
Look here #1387
you have to encode in Manchester encoding. Where 00 = 0 and 01 = F 
Example:
055151 = Code (Hex)
055151 = 0000 0101 0101 0001 0101 0001 (Binär)
00 = 0
01 = F
0000 0101 0101 0001 0101 0001 = 00FFFF0FFF0F

Jason2866, 2018-03-07:
Perhaps you take a picture from the Dip switches from the remote and the power outlet Dip switches here.
We can do the calculating together. May this helps others!

Jason2866, 2018-03-07:
Sniff the codes with your arduino receiver. The results will help!

timota, 2018-03-07:
@Jason2866 Thanks for help.
will get home after work and take all needed ss and photos.

timota, 2018-03-07:
Hi @Jason2866, 
below all requested information.

Socket: [https://www.maplin.co.uk/p/remote-controlled-mains-sockets-5-pack-n38hn](url)
RCSwitch Library: [https://github.com/sui77/rc-switch](url)
Outlet Description: [https://github.com/sui77/rc-switch/wiki/HowTo_OperateLowCostOutlets](url)  Type B: Two rotary/sliding switches
Example Sketch that used to decode: ReceiveDemo_Advanced

Photos:
[image]
[image]

Results taken at the same time for RCSwitch (Arduino UNO) and Sonoff Bridge
Button 4(ON) - 10 times
RCSwitch:

> Decimal: 1397845 (24Bit) Binary: 000101010101010001010101 Tri-State: 0FFFFFF0FFFF PulseLength: 435 microseconds Protocol: 1
> Raw data: 13500,384,1352,380,1352,380,1376,1224,512,356,1376,1228,468,20,76,540,844,64,28,368,240,748,96,872,700,192,132,180,792,180,44,2000,288,1436,32,152,512,660,1460,32,212,172,148,56,20,520,28,72,808,
> 
> Decimal: 1376256 (24Bit) Binary: 000101010000000000000000 Tri-State: 0FFF00000000 PulseLength: 435 microseconds Protocol: 1
> Raw data: 13496,388,1348,388,1360,364,1368,1236,504,368,1372,1220,524,348,1380,1228,512,352,1368,1236,412,68,276,24,92,164,296,32,56,24,440,220,16,64,712,32,984,24,408,56,56,764,108,52,32,24,144,144,316,
> 
> Decimal: 1397845 (24Bit) Binary: 000101010101010001010101 Tri-State: 0FFFFFF0FFFF PulseLength: 435 microseconds Protocol: 1
> Raw data: 13492,392,1348,376,1368,368,1364,1240,504,364,1368,1240,496,364,1372,1232,412,440,1364,1232,520,48,432,56,188,488,76,24,704,16,428,20,188,32,908,148,56,248,52,668,40,84,324,208,20,340,36,500,108,
> 
> Decimal: 1397845 (24Bit) Binary: 000101010101010001010101 Tri-State: 0FFFFFF0FFFF PulseLength: 435 microseconds Protocol: 1
> Raw data: 13492,388,1352,384,1340,404,1344,1244,496,376,1356,1248,500,364,1368,1236,504,364,20,300,308,108,72,1776,40,164,396,64,16,296,156,1320,60,32,88,40,104,48,332,44,96,240,20,980,116,880,80,104,68,
> 
> Decimal: 1397845 (24Bit) Binary: 000101010101010001010101 Tri-State: 0FFFFFF0FFFF PulseLength: 434 microseconds Protocol: 1
> Raw data: 13492,400,1344,384,1372,356,1376,1224,508,364,1376,1232,500,56,1356,1248,488,36,84,296,48,56,352,176,344,144,140,80,172,132,408,32,88,108,604,68,720,672,16,672,476,1736,48,40,44,68,16,896,76,
> 
> Decimal: 1397845 (24Bit) Binary: 000101010101010001010101 Tri-State: 0FFFFFF0FFFF PulseLength: 435 microseconds Protocol: 1
> Raw data: 13492,396,1344,384,1360,368,1372,1232,508,360,1380,1224,504,364,1376,1228,504,372,1360,1240,496,372,1360,1244,500,368,1356,28,316,24,680,24,292,408,284,52,304,36,136,116,64,220,28,536,204,24,1320,440,640,
> 
> Decimal: 1397845 (24Bit) Binary: 000101010101010001010101 Tri-State: 0FFFFFF0FFFF PulseLength: 435 microseconds Protocol: 1
> Raw data: 13496,388,1340,400,1352,372,1368,1236,504,364,1372,1224,512,356,1376,1232,504,364,1368,128,3728,24,96,16,116,196,72,132,124,804,112,52,224,2596,420,696,44,168,792,40,168,28,612,52,612,240,56,68,16,
> 
> Decimal: 1376256 (24Bit) Binary: 000101010000000000000000 Tri-State: 0FFF00000000 PulseLength: 435 microseconds Protocol: 1
> Raw data: 13492,396,1340,396,1360,364,1372,1228,500,368,1376,1224,508,364,1368,48,500,368,1364,40,240,88,796,176,344,32,84,536,212,1836,48,128,224,620,192,248,148,24,736,112,760,288,196,636,692,96,184,72,220,
> 
> Decimal: 1397760 (24Bit) Binary: 000101010101010000000000 Tri-State: 0FFFFFF00000 PulseLength: 434 microseconds Protocol: 1
> Raw data: 13492,392,1344,388,1356,372,1368,1236,496,376,1360,1240,496,372,1364,1240,496,372,1360,44,3804,56,96,284,3576,452,844,68,224,1860,220,68,576,128,1228,340,96,136,28,244,292,20,172,16,140,60,56,484,40,
> 
> Decimal: 1397845 (24Bit) Binary: 000101010101010001010101 Tri-State: 0FFFFFF0FFFF PulseLength: 435 microseconds Protocol: 1
> Raw data: 13492,392,1344,392,1344,392,1348,1244,500,368,1372,1236,492,52,52,16,104,200,1628,40,16,192,20,268,40,216,304,160,220,216,192,140,180,212,88,156,96,160,52,792,224,1700,28,44,40,28,304,52,288,

SonoffBridge:

> 23:20:45 RSL: RESULT = {"RfReceived":{"Sync":13500,"Low":430,"High":1300,"Data":"155457","RfKey":"None"}}
> 23:20:48 RSL: RESULT = {"RfReceived":{"Sync":13500,"Low":430,"High":1300,"Data":"155457","RfKey":"None"}}
> 23:20:50 RSL: RESULT = {"RfReceived":{"Sync":13490,"Low":430,"High":1300,"Data":"155557","RfKey":"None"}}
> 23:20:52 RSL: RESULT = {"RfReceived":{"Sync":13490,"Low":430,"High":1300,"Data":"155457","RfKey":"None"}}
> 23:20:55 RSL: RESULT = {"RfReceived":{"Sync":13440,"Low":440,"High":1290,"Data":"155455","RfKey":"None"}}
> 23:20:55 RSL: RESULT = {"RfReceived":{"Sync":13500,"Low":430,"High":1300,"Data":"155457","RfKey":"None"}}
> 23:20:57 RSL: RESULT = {"RfReceived":{"Sync":13500,"Low":430,"High":1300,"Data":"155557","RfKey":"None"}}
> 23:20:59 RSL: RESULT = {"RfReceived":{"Sync":13490,"Low":430,"High":1300,"Data":"155557","RfKey":"None"}}
> 23:21:02 RSL: RESULT = {"RfReceived":{"Sync":13500,"Low":430,"High":1300,"Data":"155557","RfKey":"None"}}
> 23:21:04 RSL: RESULT = {"RfReceived":{"Sync":13500,"Low":430,"High":1300,"Data":"155557","RfKey":"None"}}
> 23:21:06 RSL: RESULT = {"RfReceived":{"Sync":13500,"Low":430,"High":1300,"Data":"155557","RfKey":"None"}}

Button 4(OFF) - 10 times
RCSwitch:

> Decimal: 1397847 (24Bit) Binary: 000101010101010001010111 Tri-State: 0FFFFFF0FFF1 PulseLength: 435 microseconds Protocol: 1
> Raw data: 13488,392,1344,388,1368,356,1380,1224,508,364,1380,1216,512,360,1376,1228,508,360,1368,40,500,368,824,64,1428,16,72,356,124,172,16,272,68,104,216,32,176,60,72,248,16,176,136,92,92,540,64,24,936,
> 
[... cut ...]
> 23:22:39 RSL: RESULT = {"RfReceived":{"Sync":13500,"Low":430,"High":1300,"Data":"155557","RfKey":"None"}}
> 23:22:41 RSL: RESULT = {"RfReceived":{"Sync":13490,"Low":430,"High":1300,"Data":"155557","RfKey":"None"}}
> 23:22:43 RSL: RESULT = {"RfReceived":{"Sync":13500,"Low":430,"High":1300,"Data":"155557","RfKey":"None"}}
> 23:22:45 RSL: RESULT = {"RfReceived":{"Sync":13500,"Low":430,"High":1300,"Data":"155557","RfKey":"None"}}

Hope i gathered all needed information to do a calculation. 
Thanks

johoberm, 2018-03-08:
With your sniffed data it should work with these hex value for ON = 155455 and for OFF = 155454
Explanation: 
You have the Switch know as FLS 100
https://wiki.fhem.de/wiki/Intertechno_Code_Berechnung
(good german site about 433 rf Switches, scroll down and you will see your Picture, with the values for
the rotary DIP Switche)
This meets perfect what you haved sniffed:
ON = DEC 1397845  => HEX 155455 
0001 0101 0101 0100 0101 0101  (Binär see in sniffed data too!) 
   1	   5	    5	      4	      5       5     (hex)
0F       FF     FF      F0     FF     FF     (Manchester = DIP Switch, 0=OFF F=ON; see Website fhem)

OFF = DEC 1397844 => HEX 155454

Long in Short: you have only to convert your sniffed DEC data in HEX and send this via the RF-Bridge.
Use the timings from the RF-Bridge. These values doesnt need to match exact...

This can be defined as a Switch item (sonoff-8E8329 = RF-Bridge -> you have to change!):

Switch WohnFluter       "Wohnz. Fluter"         <light> (LR,gLight)     [ "Lighting" ] 
		{ mqtt=">[broker:cmnd/sonoff-8E8329/Backlog:command:ON:RfSync 13500; RfLow 430; RfHigh 1300; RfCode #155455],
                >[broker:cmnd/sonoff-8E8329/Backlog:command:OFF:RfSync 13500; RfLow 430; RfHigh 1300; RfCode #155454]" }

It is tagged with  [ "Lighting" ]  With that you can switch it with ALEXA by voice :-)

I would be he happy to get a feedback, and close this issue if it works

timota, 2018-03-08:
@johoberm - great, thanks. it works as expected. 

unfortunately bridge cant catch correct remote signal., so cant be used to sync state in HA. 
Anyway, it works via bridge. 

Many thanks all for your help.

Jason2866, 2018-03-08:
You are welcome!
Yes i am unhappy too, that the state sync isnt possible. If i get my second RF-Bridge i will try the alternative Firmware for the RF Chip in the RF Bridge. Maybe this works better...

timota, 2018-03-09:
Hi @Jason2866 , 

hm... didnt try. Will try today and post my result. Also i have a second remote (the same, but maybe it works in a different way).

Also, you mentioned alternative Firmware for the RF Chip. Where i can find it, what the benefit to use it? 
My RF Bridge still not fully integrated into HA, so i have a chance to play with.

Regards.

Jason2866, 2018-03-09:
Hi timota,
talk about alternative firmware here:
#1989

jcisio, 2020-12-14:
Hi,

This issue is close but I have the same problem. Well, more or less the same, because my remote has several buttons (first 4 groups for 4 sockets, and the 5th to control all of them). But different buttons are sending the same codes (or at least it's what I think).

Here is the log:
```
### ON 1
09:15:29 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:15:29","RfReceived":{"Sync":5530,"Low":200,"High":540,"Data":"555702","RfKey":"None"}}
09:15:30 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:15:30","RfReceived":{"Sync":5540,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
09:15:33 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:15:33","RfReceived":{"Sync":5540,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
09:15:35 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:15:35","RfReceived":{"Sync":5540,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
09:15:38 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:15:38","RfReceived":{"Sync":5540,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
09:15:40 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:15:40","RfReceived":{"Sync":5530,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
09:15:42 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:15:42","RfReceived":{"Sync":5540,"Low":200,"High":550,"Data":"555700","RfKey":"None"}}
09:15:45 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:15:44","RfReceived":{"Sync":5540,"Low":200,"High":540,"Data":"555500","RfKey":"None"}}
09:15:50 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:15:50","RfReceived":{"Sync":5530,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
09:15:54 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:15:54","RfReceived":{"Sync":5530,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
### ON 2
09:16:02 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:02","RfReceived":{"Sync":5530,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
09:16:06 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:06","RfReceived":{"Sync":5540,"Low":200,"High":550,"Data":"555D02","RfKey":"None"}}
09:16:07 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:07","RfReceived":{"Sync":5540,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
09:16:10 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:10","RfReceived":{"Sync":5540,"Low":200,"High":550,"Data":"555D00","RfKey":"None"}}
09:16:12 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:12","RfReceived":{"Sync":5520,"Low":190,"High":550,"Data":"555D03","RfKey":"None"}}
09:16:15 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:15","RfReceived":{"Sync":5510,"Low":190,"High":550,"Data":"555D03","RfKey":"None"}}
09:16:17 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:17","RfReceived":{"Sync":5540,"Low":200,"High":540,"Data":"555500","RfKey":"None"}}
09:16:20 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:20","RfReceived":{"Sync":5540,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
09:16:23 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:23","RfReceived":{"Sync":5540,"Low":210,"High":540,"Data":"555D02","RfKey":"None"}}
09:16:27 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:27","RfReceived":{"Sync":5530,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
### OFF 1
09:16:39 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:39","RfReceived":{"Sync":5540,"Low":200,"High":540,"Data":"555500","RfKey":"None"}}
09:16:43 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:43","RfReceived":{"Sync":5530,"Low":190,"High":550,"Data":"55570C","RfKey":"None"}}
09:16:45 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:45","RfReceived":{"Sync":5530,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
09:16:48 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:48","RfReceived":{"Sync":5520,"Low":190,"High":550,"Data":"55570C","RfKey":"None"}}
09:16:50 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:50","RfReceived":{"Sync":5530,"Low":190,"High":550,"Data":"55570C","RfKey":"None"}}
09:16:53 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:53","RfReceived":{"Sync":5530,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
09:16:55 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:55","RfReceived":{"Sync":5530,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
09:16:57 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:57","RfReceived":{"Sync":5550,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
09:16:59 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:16:59","RfReceived":{"Sync":5530,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
09:17:02 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:17:02","RfReceived":{"Sync":5530,"Low":200,"High":540,"Data":"555500","RfKey":"None"}}
### OFF 2
09:17:12 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:17:12","RfReceived":{"Sync":5550,"Low":190,"High":550,"Data":"555500","RfKey":"None"}}
09:17:14 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:17:14","RfReceived":{"Sync":5550,"Low":190,"High":550,"Data":"555D00","RfKey":"None"}}
09:17:16 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:17:16","RfReceived":{"Sync":5530,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
09:17:18 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:17:18","RfReceived":{"Sync":5520,"Low":200,"High":550,"Data":"555D0C","RfKey":"None"}}
09:17:24 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:17:24","RfReceived":{"Sync":5520,"Low":190,"High":550,"Data":"555D0C","RfKey":"None"}}
[... cut ...]
09:17:33 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:17:33","RfReceived":{"Sync":5540,"Low":200,"High":550,"Data":"555500","RfKey":"None"}}
09:17:36 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:17:36","RfReceived":{"Sync":5530,"Low":200,"High":550,"Data":"555D08","RfKey":"None"}}
09:17:39 MQT: tele/rfbridge/RESULT = {"Time":"2020-12-14T09:17:39","RfReceived":{"Sync":5550,"Low":200,"High":550,"Data":"555D00","RfKey":"None"}}
```
So, for example, the code 555500 is sent with every button.

jcisio, 2020-12-14:
@Jason2866 I reread your comment and I don't understand.

> This meets perfect what you haved sniffed:
> ON = DEC 1397845 => HEX 155455
> OFF = DEC 1397844 => HEX 155454

The value `1397845` appeared 7 times in RCSwitch but `155455` only once with RF bridge. Does that mean that what is received by the RF bridge is not precise enough? But it is still possible to use it to emit the correct codes? I don't have a 433 MHz receiver right now, is it possible to guess the code from what is received by the RF bridge?

I am really confused. The manual on/off button in the outlet could help?

Jason2866, 2020-12-14:
You can send all the codes you have. If the socket reacts it is the correct one.

jcisio, 2020-12-14:
I haven't tried it, because I have the same code for two sockets. E.g. in the above log, the code `555500` is received (but not always) when I pressed any button (ON 1 or ON 2 or OFF 1 or OFF 2). But with the comment I think I figured out: even I cannot sniff the code from the remote, I can still send the code. Is it what you meant by sync state?

jcisio, 2020-12-14:
Well, the very first comment helped: it turned out that learn code is actually working. I don't understand why, because codes are really variable. But RfKeyN work, and I have 16 keys (8 sockets) which are more than enough for me. I managed to create some automations in Home Assistant, no state sync, no button to turn on/off, but it's OK for what I need.

jcisio, 2020-12-15:
Hi,

Stop by here to let you know that I finally managed to do what I wanted. There is no state sync with the remote, but it sync with the virtual switch in Home Assistant. Here is my setup (simplified):

- I have one socket 1. I have RfKey1 and RfKey2 learnt to turn off/on the socket. I have these rules in Tasmota so that there is a MQTT topic to control and get the states:
```
    rule1
        on RfKey1 do Publish2 rfbridge/switch1 OFF break
        on Event#switch1off do Publish cmnd/rfbridge/RfKey1 endon
        on RfKey2 do Publish2 rfbridge/switch1 ON break
        on Event#switch1on do Publish cmnd/rfbridge/RfKey2 endon
```
- Then in Home Assistant, I can create a new switch with state sync:
```
switch:
  - platform: mqtt
    name: "Switch 1"
    state_topic: "rfbridge/switch1"
    command_topic: "cmnd/rfbridge/event"
    payload_on: "switch1on"
    payload_off: "switch1off"
    state_on: "ON"
    state_off: "OFF"
```
Now everything works as expected. Well, except that the RF is not reliable and has no feedback, e.g. every once in a while the socket does not receive the code. But it's not ofter, like 1/20 of the time, only with the one in the front yard (10 meters and a few walls), no problem with the one in the living room. Again, well, it's just RF, I work around it by sending the code multiple times.

kiat-huang, 2021-09-24:
I have a Sonoff RM433 keypad (8 keys) and want to get Home Assistant to recognize the state and change of state of the buttons.  I also want to get my car's inbuilt garage door control to be able to clone some of the buttons so I can configure Home Assistant to control various entities.  As yet I have no RF clients.  Similar to @jcisio and based on his config, I got it working with the RF bridge with the following.

_Commands on the RF bridge._

> rule1 
   on RfReceived#RfKey==1 do Publish2 sonoff/rfbridge/switch1 ON break 
   on Event#switch1on do Publish cmnd/sonoff/rfbridge/RfKey1 endon 
   on RfReceived#RfKey==5 do Publish2 sonoff/rfbridge/switch1 OFF break
   on Event#switch1off do Publish cmnd/sonoff/rfbridge/RfKey5 endon
> 
>   on RfReceived#RfKey==2 do Publish2 sonoff/rfbridge/switch2 ON break 
   on Event#switch2on do Publish cmnd/sonoff/rfbridge/RfKey2 endon
   on RfReceived#RfKey==6 do Publish2 sonoff/rfbridge/switch2 OFF break
   on Event#switch2off do Publish cmnd/sonoff/rfbridge/RfKey6 endon
> 
>   on RfReceived#RfKey==3 do Publish2 sonoff/rfbridge/switch3 ON break
   on Event#switch3on do Publish cmnd/sonoff/rfbridge/RfKey3 endon
   on RfReceived#RfKey==7 do Publish2 sonoff/rfbridge/switch3 OFF break
   on Event#switch3off do Publish cmnd/sonoff/rfbridge/RfKey7 endon
> 
>   on RfReceived#RfKey==4 do Publish2 sonoff/rfbridge/switch4 ON break
   on Event#switch4on do Publish cmnd/sonoff/rfbridge/RfKey4 endon
   on RfReceived#RfKey==8 do Publish2 sonoff/rfbridge/switch4 OFF break
   on Event#switch4off do Publish cmnd/sonoff/rfbridge/RfKey8 endon
>
> rule1 1

_and in Home Assistant in switches.yaml_

> - platform: mqtt
  name: "RF Switch 1"
  state_topic: "sonoff/rfbridge/switch1"
  command_topic: "sonoff/rfbridge/switch1"
  payload_on: "ON"
  payload_off: "OFF"
  retain: true
> 
> - platform: mqtt
  name: "RF Switch 2"
  state_topic: "sonoff/rfbridge/switch2"
  command_topic: "sonoff/rfbridge/switch2"
  payload_on: "ON"
  payload_off: "OFF"
  retain: true
> 
> - platform: mqtt
  name: "RF Switch 3"
  state_topic: "sonoff/rfbridge/switch3"
  command_topic: "sonoff/rfbridge/switch3"
  payload_on: "ON"
  payload_off: "OFF"
  retain: true
> 
> - platform: mqtt
  name: "RF Switch 4"
  state_topic: "sonoff/rfbridge/switch4"
  command_topic: "sonoff/rfbridge/switch4"
  payload_on: "ON"
  payload_off: "OFF"
  retain: true