arendst/Tasmota #2099: Error 255 - Tasmota

Greengeeks, 2018-03-07:
Hello,
what I have/do:

- MacBook Air 11' (macOS High Sierra)

- Sonoff Basic

- Arduino

- Atom

- Press and hold "Wi-Fi" button when I plug in computer

I tried switching TX and RX but nothing.
The problem is always the same, like other users.

So:

1. Can be the problem the Mac?

2. Can I load Tasmota on Arduino and so on Sonoff?

Thanks!

[image]

ZayedSchroeder, 2018-03-07:
Hi Guys, please help with the above, I have tried it with a Genuine Arduino Uno R3, Duinotech Nano V3.0, FTDI with FT232 Chip, a standard USB to Serial cable i use to configure my Cisco switches and even tried a Raspberry Pi Zero W 1.1 and still had no luck having the same error as above. I tried two different laptops with Win20, one laptop with Win 7 and also a Macbook pro, still no luck.
On the 4CH Pro i believe i have to pin GPIO0 to ground which is what i did. Connected TX to RX, VCC and ground. Swapped over TX to RX with no change. I have used Arduino IDE, Atom (Platformio) ESPEasy but bam still same result.

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
* [upload] Error -1
 [ERROR] Took 36.32 seconds

To setup in program mode i used the link below
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-4CH-and-4CH-Pro
Does LED 1,2 and 3 light up for programming mode? Also went through "Solving Sonoff 4CHpro programing issues"

I am now at a point where i don't even need to jumper GPO0 and Ground as it starts up with the 3 LEDS on.
I am definitely using the correct COM ports as whenever i change the FTDI or USB converter i would check in device manager to see what port it is set to.
I finally went out and bought a Google Home as EWELink integrates with that, now i cant even get it back to its original status.
Can anybody please give me some pointers either way, with Tasmota upgrade or back to original status.
Thanks in advance
Kind regards
Zayed

issacg, 2018-03-08:
FWIW, unless I'm actually debugging the firmware, I *always* use [SonOTA](https://github.com/mirko/SonOTA) to flash.  If you're still on the stock firmware, I highly recommend using that.

@ZayedSchroeder note that for devices other than Sonoff that are based on the ESP chipset, you may need to adjust the settings in platformio.ini to match your hardware.  Different hardware sometimes need different upload settings.  For example, I often hack on a NodeMCU and use the following

```ini
board = nodemcuv2
board_flash_mode = qio
upload_resetmethod = nodemcu
```

Frogmore42, 2018-03-08:
Looking at your provided screenshot, you are using an older version and you have not selected a single configuration to download.  So, it is building all 4 and trying to download them all.  The 1st one succeeded, but the rest did not.  Uncomment out the one configuration you want and try again.  You are almost there.

ZayedSchroeder, 2018-03-12:
issacg, u r da man !! thank you very much for the SonOta option, i did have a look at that and thought my device was purchased late 2017 and was under the impression that the firmware was above 1.6 and i did not even try that path. After your comment i decided to five it a go and VOILA !! i am up and running. Thank you very much.
Now i have 4 of the 4ch pro's and stuffed 2 of them already, does anybody know how to fix them as i only get LED 1,2, & 3 on as soon as i apply power. It is almost as if it is stuck on firmware upgrade mode and i then am unable to see the ITEAD.XXXX AP, to complete to SonOTA upgrade.

Thank again for your help :) and for any further help from anyone :)
Kind regards

Zayed