arendst/Tasmota #2100: Sonoff behaviour changes when mqtt disconnects

awaldeck, 2018-03-07:
Hi,

I have a problem with a sonoff running tasmota.

The sonoff controls something with pulsetime using the relay. Additionally 2 of the GPIO's are used to monitor the status of 2 reed switches and send the status over mqtt with switchtopic
Now this works perfectly however if the mqtt server is down for whatever reason, the sonoff starts using the reed switches as "buttons" to trigger the relay causing a very undesirable operation.

Is there some way to keep the config working as is when mqtt is down for whatever reason, failure, maintenance, acts of god... etc.. ?

Hardware: [Sonoff SV]
Status 0:
- [17:13:06 MQT: stat/door/STATUS = {"Status":{"Module":3,"FriendlyName":"door","Topic":"door","ButtonTopic":"door","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":1,"PowerRetain":0}}
17:13:06 MQT: stat/door/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://x.x.x.x/tasmota/sonoff.bin","Uptime":"0T00:03:12","Sleep":0,"BootCount":52,"SaveCount":321,"SaveAddress":"F6000"}}
17:13:06 MQT: stat/door/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-03-07T11:12:21","Boot":4,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
17:13:06 MQT: stat/door/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"syslog","LogPort":514,"SSId1":"mysecretwifi","SSId2":"secondsecretwifi","TelePeriod":300,"SetOption":"00000049"}}
17:13:06 MQT: stat/door/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
17:13:06 MQT: stat/door/STATUS5 = {"StatusNET":{"Hostname":"door-3868","IPAddress":"x.x.x.x","Gateway":"x.x.x.x","Subnetmask":"x.x.x.x","DNSServer":"x.x.x.x","Mac":"x.x.x.x.x.x.x.x","Webserver":2,"WifiConfig":4}}
17:13:06 MQT: stat/door/STATUS6 = {"StatusMQT":{"MqttHost":"x.x.x.x","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_202F1C","MqttUser":"xxxx","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:13:06 MQT: stat/door/STATUS7 = {"StatusTIM":{"UTC":"Wed Mar 07 15:13:06 2018","Local":"Wed Mar 07 17:13:06 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2}}
17:13:06 MQT: stat/door/STATUS10 = {"StatusSNS":{"Time":"2018-03-07T17:13:06","Switch2":"OFF","Switch3":"OFF"}}
17:13:06 MQT: stat/door/STATUS11 = {"StatusSTS":{"Time":"2018-03-07T17:13:06","Uptime":"0T00:03:12","Vcc":3.618,"POWER":"OFF","Wifi":{"AP":2,"SSId":"xxxx","RSSI":100,"APMac":"9C:97:26:DD:24:FF"}}} ] Provide the output of command ``status 0``

awaldeck, 2018-03-13:
I have tried every combination of switchtopic, switchmode, etc. When MQTT Server is up everything works 100% Relay is toggled with mqtt commands, the GPIO inputs specified is not to toggled but only sends MQTT topics.
However the moment MQTT Server is down the GPIO starts triggering the relay (only happens when no connction to MQTT can be made)

karatefeet, 2018-05-10:
Having the same issue with a sonoff SV. Following this in hopes of finding a solution. Since I have my reed switch to close when the garage door is down, this issue causes it to stay open so not an ideal situation.

JohnnyPicnic, 2018-05-21:
I too have the same problem. I thought setting switchmode2 in my case would solve the problem but it still persists. Only happens when Tasmota is unable to connect to MQTT server.

JohnnyPicnic, 2018-05-24:
Just tried with latest firmware and the issue still persists. 

04:12:20 RSL: Group 0, Index 1, Command STATUS, Data 0
04:12:20 MQT: stat/garage/STATUS = {"Status":{"Module":3,"FriendlyName":["garageIO"],"Topic":"garage","ButtonTopic":"garage","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
04:12:20 MQT: stat/garage/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:16:43","StartupUTC":"2018-05-24T02:55:37","Sleep":0,"BootCount":45,"SaveCount":124,"SaveAddress":"F8000"}}
04:12:20 MQT: stat/garage/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-05-15T15:29:54","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
04:12:20 MQT: stat/garage/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["DTRON","indebuurt2"],"TelePeriod":300,"SetOption":["558000A9","55818000"]}}
04:12:20 MQT: stat/garage/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
04:12:20 MQT: stat/garage/STATUS5 = {"StatusNET":{"Hostname":"GarageIO","IPAddress":"192.168.1.30","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:95:9B:9F","Webserver":2,"WifiConfig":5}}
04:12:20 MQT: stat/garage/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.9","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_959B9F","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
04:12:21 MQT: stat/garage/STATUS7 = {"StatusTIM":{"UTC":"Thu May 24 03:12:20 2018","Local":"Thu May 24 04:12:20 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:57","Sunset":"20:36"}}
04:12:21 MQT: stat/garage/STATUS10 = {"StatusSNS":{"Time":"2018-05-24T04:12:21","Switch2":"OFF"}}

JohnnyPicnic, 2018-05-24:
There is something fishy happening with switchmode 2. When I use swtichmode 1 it works as expected with no mqtt server.

karatefeet, 2018-05-25:
@JohnnyPicnic Give this a try. It worked great for me with a Sonoff SV. Relay does not activate if the MQTT / HA connection is lost. https://www.youtube.com/watch?v=QMepwpyjMCY&lc=UgxIjzN_RzWL5vU2GQJ4AaABAg.8blrXJl8ffy8bvfeEqB4sC

If it doesn't link to the right comment, do a search for "Amazing work". Kudos to Pierre.

JohnnyPicnic, 2018-05-25:
Yes that's the video I used to get up and running. Could you be more specific on the steps you used to solve the issue?

karatefeet, 2018-05-25:
A commenter by the name of Pierre figured out a solution in the comments which I linked to.
Here’s the gist:
Switchmode1 0 and Switchmode2 2 does not seem to isolate GPIO14 from the relay on the Sonoff Basic.
Setting GPIO3 to relay2 as well does the trick.
The Tasmota interface now shows 2 Toggle buttons, One for the relay/LED and one for GPIO14/LED.

Also, setting pulsetime1 10, turns off the relay after 1 second, possibly eliminating the need for the automation in HA.
BTW set PowerOnState 0 to prevent activation after a power interruption.

1. Configure Sonoff Basic with GPIO3 - 18 Relay2 and GPIO14 - 10 Switch2.
2. pulsetime1 10 (Switches off relay after 1 second)
3. PowerOnState 0 (prevents activation on electrical power up)
4. Switchtopic2 DoorStatus -> changes from stat to cmnd
5. switchmode 2 2 -> changes TOGGLE to ON / OFF
6. switchmode 1 0 needed? -> I didn’t set this

Again, this solution is working great for me.

alfredopironti, 2018-06-03:
I'm having the same issue.

In my configuration, SwitchTopic is set to a String. When the Sonoff (basic) can connect to an MQTT server, triggering the switch issues a 'cmnd' over the defined SwitchTopic (which is the expected behavior).

However, if the connection to the MQTT server is lost, triggering the switch directly controls the relay, as if SwitchTopic was set to 0. (Crucially, I use SwitchMode<x> set to 2 (Follow), and it correctly turns the relay on and off according to the ON/OFF state reported by the switch.)

I believe SwitchTopic configuration is not honored when the MQTT host is down, which looks like a software flaw.

closed 2018-08-02 by stale[bot]