arendst/Tasmota #2103: Client already connected, closing old connection

codenamej6, 2018-03-07:
I'm having some trouble adding some new switches to hass.io.
I've already successfully added 6 sonoff basics so I don't think that I'm doing anything wrong or different than those first 6, but every other one after that has been giving me different problems.  I've tried four different switches and they either timeout after 15 seconds or they give me an "already connected, closing old connection" message.
I've restarted hass.io multiple times, uninstalled/reinstalled mqtt, tried different names, different outlets, and they all act the same way.

I'd appreciate any advice to what could be causing this.  

Log from moquitto:

starting version 3.2.2
1520454504: mosquitto version 1.4.12 (build date 2017-06-01 13:03:48+0000) starting
1520454504: Config loaded from /etc/mosquitto.conf.
1520454504: Opening ipv4 listen socket on port 1883.
1520454504: Opening ipv6 listen socket on port 1883.
1520454504: Warning: Mosquitto should not be run as root/administrator.
1520454505: New connection from 172.30.32.1 on port 1883.
1520454506: Socket error on client <unknown>, disconnecting.
1520454506: New connection from 172.30.32.1 on port 1883.
1520454506: New client connected from 172.30.32.1 as c37b7bdd-4197-4fc2-a115-1fae716a9004 (c1, k60, u'J6').
1520454512: New connection from 192.168.0.113 on port 1883.
1520454512: New client connected from 192.168.0.113 as TasmotaTest01 (c1, k15, u'J6').
1520454512: New connection from 192.168.0.114 on port 1883.
1520454512: New client connected from 192.168.0.114 as DVES_0012D4 (c1, k15, u'J6').
1520454512: New connection from 192.168.0.103 on port 1883.
1520454512: New client connected from 192.168.0.103 as LivingRoomLight (c1, k15, u'J6').
1520454513: New connection from 192.168.0.100 on port 1883.
1520454513: New client connected from 192.168.0.100 as DreamScreenPower (c1, k15, u'J6').
1520454513: New connection from 192.168.0.101 on port 1883.
1520454513: New client connected from 192.168.0.101 as KitchenCabinetLight (c1, k15, u'J6').
1520454513: New connection from 192.168.0.123 on port 1883.
1520454513: New client connected from 192.168.0.123 as BasementHallwayLightTest (c1, k15, u'J6').
1520454534: New connection from 192.168.0.123 on port 1883.
1520454534: Client BasementHallwayLightTest already connected, closing old connection.
1520454534: Client BasementHallwayLightTest disconnected.
1520454534: New client connected from 192.168.0.123 as BasementHallwayLightTest (c1, k15, u'J6').
1520454554: New connection from 192.168.0.123 on port 1883.
1520454554: Client BasementHallwayLightTest already connected, closing old connection.
1520454554: Client BasementHallwayLightTest disconnected.
1520454554: New client connected from 192.168.0.123 as BasementHallwayLightTest (c1, k15, u'J6').
1520454585: New connection from 192.168.0.123 on port 1883.
1520454585: Client BasementHallwayLightTest already connected, closing old connection.
1520454585: Client BasementHallwayLightTest disconnected.
1520454585: New client connected from 192.168.0.123 as BasementHallwayLightTest (c1, k15, u'J6').
1520454609: New connection from 192.168.0.123 on port 1883.
1520454609: Client BasementHallwayLightTest already connected, closing old connection.

dzinyit, 2018-03-24:
I'm having the same issue.

plastik30, 2018-05-10:
Same issue here..

1525953658: New client connected from 192.168.0.153 as sonoff2 (c1, k15, u'george').
1525953675: Socket error on client sonoff2, disconnecting.
1525953675: New connection from 192.168.0.153 on port 1883.
1525953675: New client connected from 192.168.0.153 as sonoff2 (c1, k15, u'george').
1525953737: New connection from 192.168.0.153 on port 1883.
1525953737: Client sonoff2 already connected, closing old connection.
1525953737: Client sonoff2 disconnected.
1525953737: New client connected from 192.168.0.153 as sonoff2 (c1, k15, u'george').

codenamej6, 2018-05-10:
Someone suggested that I try something and it solved the problem for me immediately without any problems.  I re-flashed it but in the user_config section, I commented out:
#define USE_DISCOVERY

plastik30, 2018-05-11:
I've commented the " #define USE_DISCOVERY " line and now everything is ok. Thanks !

closed 2018-06-02 by ascillato

tenkaibr, 2020-08-10:
same problem here