arendst/Tasmota #2104: Unable to probe wemo at 192.168.1.xxx

vpomax, 2018-03-07:
- Raspberry Pi3 
- Home Assistant 0.64.3
- HASSIO v. 0.96
- 8 devices Sonoff Touch succesfully flashed with Tasmota fw 5.12.0 

At reboot I receive these messages in log file...

2018-03-07 23:25:30 WARNING (MainThread) [homeassistant.setup] Setup of wemo is taking over 10 seconds.
2018-03-07 23:25:33 WARNING (SyncWorker_0) [homeassistant.components.wemo] Unable to probe wemo at 192.168.1.87
2018-03-07 23:25:34 WARNING (SyncWorker_0) [homeassistant.components.wemo] Unable to probe wemo at 192.168.1.86
2018-03-07 23:25:34 WARNING (SyncWorker_0) [homeassistant.components.wemo] Unable to probe wemo at 192.168.1.91
2018-03-07 23:25:34 WARNING (SyncWorker_0) [homeassistant.components.wemo] Unable to probe wemo at 192.168.1.84
2018-03-07 23:25:34 WARNING (SyncWorker_0) [homeassistant.components.wemo] Unable to probe wemo at 192.168.1.85
2018-03-07 23:25:44 WARNING (SyncWorker_0) [homeassistant.components.wemo] Unable to probe wemo at 192.168.1.81
2018-03-07 23:25:44 WARNING (SyncWorker_0) [homeassistant.components.wemo] Unable to probe wemo at 192.168.1.82
2018-03-07 23:25:44 WARNING (SyncWorker_0) [homeassistant.components.wemo] Unable to probe wemo at 192.168.1.83
2018-03-07 23:25:44 WARNING (SyncWorker_0) [homeassistant.components.wemo] Unable to probe wemo at 192.168.1.84
2018-03-07 23:25:44 WARNING (SyncWorker_0) [homeassistant.components.wemo] Unable to probe wemo at 192.168.1.85
2018-03-07 23:25:45 WARNING (SyncWorker_0) [homeassistant.components.wemo] Unable to probe wemo at 192.168.1.86
2018-03-07 23:25:45 WARNING (SyncWorker_0) [homeassistant.components.wemo] Unable to probe wemo at 192.168.1.87
2018-03-07 23:25:45 WARNING (SyncWorker_0) [homeassistant.components.wemo] Unable to probe wemo at 192.168.1.91

mattlward, 2018-03-09:
I have a ticket open for basically the same issue.  With wemo emulation turned off they still appear to HA and Alexa as viable wemo devices.

vpomax, 2018-03-09:
Thank you mattlward 
I turned off Belkin Wemo emulation. now I haven't errors and all Sonoff flashed with Tasmota works fine!

mattlward, 2018-03-12:
I tried that and one unit is hanging on to emulation even when turned off.  I hope they fix this, causes issues with Alexa and errors with HA.

GasTurbineMan, 2018-05-22:
Not wanting to hi-jack your post, but I have a general question about WeMo Emulation and linking to Alexa?  I cannot get Alexa to detect my Tasmota flashed Sonoffs emulating WeMo.  And I cannot add the WeMo skill to Alexa because I don't have a WeMo account to link to.

What skill is needed by Alexa to detect the WeMo emulated Tasmota Sonoffs?

Thanks

closed 2018-07-10 by ascillato2

tlpbu, 2018-08-30:
Why is this ticket closed with no solution whatsoever.
Still happens regulary
`2018-08-30 19:53:56 WARNING (SyncWorker_4) [homeassistant.components.wemo] Unable to probe wemo at 192.168.0.21
2018-08-30 19:53:57 WARNING (SyncWorker_4) [homeassistant.components.wemo] Unable to probe wemo at 192.168.0.30
2018-08-30 19:53:57 WARNING (SyncWorker_4) [homeassistant.components.wemo] Unable to probe wemo at 192.168.0.27
2018-08-30 19:53:58 WARNING (SyncWorker_4) [homeassistant.components.wemo] Unable to probe wemo at 192.168.0.28
2018-08-30 19:53:59 WARNING (SyncWorker_4) [homeassistant.components.wemo] Unable to probe wemo at 192.168.0.29`

ascillato, 2018-08-30:
Please, search for wifi issues for troubleshoting and upgrade to last Tasmota version.