arendst/Tasmota #2105: sonoff T1 relay led intensity

dragouf, 2018-03-07:
I use a sonoff T1 3gang uk. I would like to know if it's possible to change the relay led intensity (circle one) when relay is off ?

I have 2 different version of the T1 switch, one is very bright when off and the other one is barely visible.

I prefer when it's brighter to be able to see the switch in the night.

Is it possible to change this by changing tasmota code or even changing something physically ?

thanks,

arendst, 2018-03-08:
You could try to configure the led gpio as a pwm channel and control it's intensity that way. You will loose the functionality of reverse power on/off though...

dragouf, 2018-03-08:
Ah great, so you mean the relay led is wired to a gpio of the esp chip ? Where I can find with one ?

arendst, 2018-03-08:
Only the wifi led is directly connected to a gpio. The relay leds can not be controlled.

dragouf, 2018-03-08:
@arendst  ah so in finaly there is no luck to control it ?
I saw an issue about that about sonoff touch, so it's same for T1 ?

gonzunigad, 2018-09-09:
Just to confirm... there is no way to control relay leds on sonoff T1?

ascillato, 2018-09-09:
Hi, not by software. They are hardwired to the relays. If you don't want them, just desolder them.