arendst/Tasmota #2109: syslog stops working after reboot

cyablo, 2018-03-08:
I'm using one of my Debian Machines as remote syslog Server. If have setup all of my tasmota devices to syslog to this remote Machine. Remote syslog works fine till I reboot one of my tasmota's. After this, the the rebootet device stop logging to the remote machine until I go into the Logging options and simply press "save". After saving it start logging to remote again.

curzon01, 2018-03-08:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``

cyablo, 2018-03-08:
Added Info.

arendst, 2018-03-08:
Status output looks fine. Pls try 5.12.0d which has some changes regarding syslog DNS.

bollitec, 2018-05-03:
Hello there,

found this issue with nearly the same situation:

Workflow:
a) Set via MQTT SysLogInfo: mosquitto_pub -t 'cmnd/sonoffs/Backlog' -m 'loghost raspi; syslog 4; logport 514'
b) Response: stat/sonoff1/RESULT {"Backlog":"Appended"}
                      stat/sonoff1/RESULT {"LogHost":"raspi"}
                      stat/sonoff1/RESULT {"LogHost":"raspi"}
                      stat/sonoff1/RESULT {"SysLog":"4 **(Active 0)**"}
                      stat/sonoff1/RESULT {"LogPort":514}
c) no input to SysLog Server => seems the Active 0
d) got to web and click only save (as mentioned from @cyablo)...
                       mosquitto_pub -t 'cmnd/sonoffs/Backlog' -m 'syslog'
                       stat/sonoff1/RESULT {"SysLog":"4 **(Active 4)**"}
e) reboot and control
                       mosquitto_pub -t 'cmnd/sonoffs/Backlog' -m 'syslog'
                       stat/sonoff1/RESULT {"SysLog":"4 **(Active 0)**"}

How can the Syslog activated without use the web save button?

reloxx13, 2018-07-01:
this behaviour still exists.

without save active 0, after save its active 4

mrohner, 2018-08-11:
Same issue here with Tasmota 5.13.1
<09:51:33 CMD: status 0
09:51:33 MQT: stat/wemos82/STATUS = {"Status":{"Module":18,"FriendlyName":["television","blavbla"],"Topic":"wemos82","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
09:51:33 MQT: stat/wemos82/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:10:54","StartupUTC":"2018-08-11T07:40:39","Sleep":0,"BootCount":10,"SaveCount":22,"SaveAddress":"3FA000"}}
09:51:33 MQT: stat/wemos82/STATUS2 = {"StatusFWR":{"Version":"5.13.1","BuildDateTime":"2018-08-11T09:40:13","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
09:51:33 MQT: stat/wemos82/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":2,"LogHost":"192.168.178.59","LogPort":514,"SSId":["mmwireless",""],"TelePeriod":300,"SetOption":["00008009","55C18000"]}}
09:51:33 MQT: stat/wemos82/STATUS4 = {"StatusMEM":{"ProgramSize":496,"Free":524,"Heap":10,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2}}
09:51:33 MQT: stat/wemos82/STATUS5 = {"StatusNET":{"Hostname":"wemos82","IPAddress":"192.168.178.82","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"2C:3A:E8:0B:4D:60","Webserver":2,"WifiConfig":3}}
09:51:33 MQT: stat/wemos82/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.59","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0B4D60","MqttUser":"mqtt","MqttType":3,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:51:33 MQT: stat/wemos82/STATUS7 = {"StatusTIM":{"UTC":"Sat Aug 11 07:51:33 2018","Local":"Sat Aug 11 09:51:33 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
09:51:33 MQT: stat/wemos82/STATUS10 = {"StatusSNS":{"Time":"2018-08-11T09:51:33"}}
09:51:33 MQT: stat/wemos82/STATUS11 = {"StatusSTS":{"Time":"2018-08-11T09:51:33","Uptime":"0T00:10:54","Vcc":2.779,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"mmwireless","RSSI":36,"APMac":"38:10:D5:F1:C7:0C"}}}
>

mrohner, 2018-08-13:
Well, I just updated all devices to Tasmota 5.13.1. I will not install new releases just for that bug. You probably know the with new releases you get new bugs. So I will not change my running system.

ascillato2, 2018-09-04:
@cyablo 

Hi,

Have you managed to solve your issue?

cyablo, 2018-09-04:
I'm not 100% sure since I migrated my main VM from jessie to stretch and did not changed the syslog destination IP in tasmota since then. But a quick test shows the excact same behavior. No logging after reboot till you hit the save button again, running Tasmota 6.2.0.

curzon01, 2018-09-04:
Can't still reproduce the issue,a lso not with workflow from @bollitec  - it may be related to other cases but the infos given here are to less - what's the steps from first flash a device to reproduce the issue

arendst, 2018-09-04:
Cannot confirm.

Pls provide latest status 0 information.

cyablo, 2018-09-04:
Status now:

> 12:34:56 MQT: stat/tasmota7/STATUS = {"Status":{"Module":8,"FriendlyName":["Lichterkette"],"Topic":"tasmota7","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
12:34:56 MQT: stat/tasmota7/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://:80/sonweb/data/firmwares/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T04:19:37","StartupUTC":"2018-09-04T06:15:19","Sleep":50,"BootCount":55,"SaveCount":597,"SaveAddress":"F4000"}}
12:34:56 MQT: stat/tasmota7/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018.08.31 15:19:15","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:34:56 MQT: stat/tasmota7/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"192.168.1.26","LogPort":514,"SSId":["xxx","xxx"],"TelePeriod":300,"SetOption":["00000009","55A18000","00000001"]}}
12:34:56 MQT: stat/tasmota7/STATUS4 = {"StatusMEM":{"ProgramSize":473,"Free":528,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","0FDAE794","24000000","23B617CE","00000000"]}}
12:34:56 MQT: stat/tasmota7/STATUS5 = {"StatusNET":{"Hostname":"tasmota7","IPAddress":"192.168.1.43","Gateway":"192.168.1.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.18","Mac":"EC:FA:BC:12:F7:E5","Webserver":2,"WifiConfig":5}}
12:34:56 MQT: stat/tasmota7/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.26","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_12F7E5","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:34:56 MQT: stat/tasmota7/STATUS7 = {"StatusTIM":{"UTC":"Tue Sep 04 10:34:56 2018","Local":"Tue Sep 04 12:34:56 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:11","Sunset":"20:27"}}
12:34:56 MQT: stat/tasmota7/STATUS10 = {"StatusSNS":{"Time":"2018-09-04T12:34:56"}}
12:34:56 MQT: stat/tasmota7/STATUS11 = {"StatusSTS":{"Time":"2018-09-04T12:34:56","Uptime":"0T04:19:37","Vcc":3.196,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxx","RSSI":82,"APMac":"xxx"}}}

Status after reboot:

> 12:38:54 MQT: stat/tasmota7/STATUS = {"Status":{"Module":8,"FriendlyName":["Lichterkette"],"Topic":"tasmota7","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
12:38:54 MQT: stat/tasmota7/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://:80/sonweb/data/firmwares/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:00:17","StartupUTC":"2018-09-04T10:38:37","Sleep":50,"BootCount":57,"SaveCount":599,"SaveAddress":"FA000"}}
12:38:54 MQT: stat/tasmota7/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018.08.31 15:19:15","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:38:54 MQT: stat/tasmota7/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"192.168.1.26","LogPort":514,"SSId":["xxx","xxx"],"TelePeriod":300,"SetOption":["00000009","55A18000","00000001"]}}
12:38:54 MQT: stat/tasmota7/STATUS4 = {"StatusMEM":{"ProgramSize":473,"Free":528,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","0FDAE794","24000000","23B617CE","00000000"]}}
12:38:54 MQT: stat/tasmota7/STATUS5 = {"StatusNET":{"Hostname":"tasmota7","IPAddress":"192.168.1.43","Gateway":"192.168.1.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.18","Mac":"EC:FA:BC:12:F7:E5","Webserver":2,"WifiConfig":5}}
12:38:54 MQT: stat/tasmota7/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.26","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_12F7E5","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:38:54 MQT: stat/tasmota7/STATUS7 = {"StatusTIM":{"UTC":"Tue Sep 04 10:38:54 2018","Local":"Tue Sep 04 12:38:54 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:11","Sunset":"20:27"}}
12:38:54 MQT: stat/tasmota7/STATUS10 = {"StatusSNS":{"Time":"2018-09-04T12:38:54"}}
12:38:54 MQT: stat/tasmota7/STATUS11 = {"StatusSTS":{"Time":"2018-09-04T12:38:54","Uptime":"0T00:00:17","Vcc":3.196,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxx","RSSI":88,"APMac":"xxx"}}}

And again, no logs until save button under logging is pressed once.

arendst, 2018-09-04:
Set sleep to 0 and try again pls.

cyablo, 2018-09-04:
Set sleep to 0 -> reboot = no logs :(

ascillato2, 2018-09-07:
Hi

I have tested syslog with last version v6.2.1.2 and if I set `syslog 4` and make a software reboot or a powercycle it remains at 4.

Could be that you have a retain flag in your mqtt server that is sending `syslog 0`?

To check this, please enable `weblog 4`, make a reboot and check what are the commands coming from mqtt at startup.

ascillato, 2018-09-09:
Hi,

Have you managed to solve your issue?

reloxx13, 2018-09-09:
nope, still not fixed...

re-produce:
go on web gui
set syslog 4 more debug
hit save
check syslog => incoming logs
console enter syslog, result => 
```
19:05:30 CMD: syslog 4
19:05:30 RSL: RESULT = {"SysLog":"4 (Active 4)"}
```

restart device by command restart 1

check syslog => no new entries
go on web gui
check syslog config => still on syslog 4 more debug
enter syslog in console, result =>
```
19:07:55 CMD: syslog
19:07:55 RSL: RESULT = {"SysLog":"4 (Active 0)"}
```

go back to logging config page on web gui
hit save
check syslog => new logs incoming
checksyslog command, result =>
```
19:09:48 CMD: syslog
19:09:48 RSL: RESULT = {"SysLog":"4 (Active 4)"}
```

ascillato, 2018-09-09:
In your console, when Tasmota starts, it shows something about syslog like a retain message?

reloxx13, 2018-09-09:
no ha/mqtt in use, updated comment, pls wait a sec ill add status 0 on above command

reloxx13, 2018-09-09:
log right after restart 1
```
00:00:00 Projekt sonoff sOnOff-4-4MB (Topic sonoff, Fallback DVES_746DCF, GroupTopic sonoffs) Version 6.2.0.2-2_3_0
00:00:00 WIF: verbinden mit AP1 HomeZone in Modus 11N wie sOnOff-4-4MB...
00:00:06 WIF: verbunden
00:00:06 DNS: initialisiert
00:00:06 HTP: Web-Server aktiv bei sOnOff-4-4MB.local mit IP-Adresse 192.168.178.74
00:00:06 UPP: Multicast (wieder-)verbunden
00:00:06 RSL: INFO1 = {"Module":"Sonoff Basic","Version":"6.2.0.2","FallbackTopic":"DVES_746DCF","GroupTopic":"sonoffs"}
00:00:06 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sOnOff-4-4MB","IPAddress":"192.168.178.74"}
00:00:06 RSL: INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 RSL: RESULT = {"POWER":"OFF"}
00:00:06 RSL: POWER = OFF
```

ascillato, 2018-09-10:
Hi,

I have just configured a Syslog Server in a RPI following this [tutorial](https://www.sigmdel.ca/michel/ha/rpi/syslog_en.html) (link also in the wiki)

I could not reproduce the issue (the syslog works fine after reboot using last dev version):

The output of the console (with weblog 4):
```
00:00:00 CFG: Cargado desde flash en 3F9, Conteo 73
00:00:00 APP: Conteo Reinicios 31
00:00:00 SRC: Restart
00:00:00 Proyecto sonoff Living (Topic living, Fallback DVES_591719, GroupTopic sonoffs) Versión 6.2.1.3-2_4_2
00:00:00 CFG: Grabado a la flash en 3F8, Conteo 74, Bytes 3584
00:00:00 WIF: Probando conexión...
00:00:00 WIF: Intentando conectar...
00:00:00 WIF: Connectando a AP1 NetWireless en modo 11N como living-5913...
00:00:01 WIF: Probando conexión...
00:00:01 WIF: Intentando conectar...
00:00:02 WIF: Probando conexión...
00:00:02 WIF: Intentando conectar...
00:00:03 WIF: Probando conexión...
00:00:03 WIF: Intentando conectar...
00:00:04 WIF: Probando conexión...
00:00:04 WIF: Conectado
00:00:04 HTP: Servidor web activo en living-5913 con dirección IP 192.168.1.33
00:00:05 MQT: Intentando conectar...
00:00:07 APP: (UTC) Mon Sep 10 21:28:11 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
18:28:14 MQT: Conectado
18:28:14 MQT: tele/living/LWT = Online (Grabado)
18:28:14 MQT: cmnd/living/POWER = 
18:28:14 MQT: Suscribir a cmnd/living/#
18:28:14 MQT: Suscribir a cmnd/sonoffs/#
18:28:14 MQT: Suscribir a cmnd/DVES_591719/#
18:28:14 MQT: tele/living/INFO1 = {"Module":"Generic","Version":"6.2.1.3","FallbackTopic":"DVES_591719","GroupTopic":"sonoffs"}
18:28:14 MQT: tele/living/INFO2 = {"WebServerMode":"Admin","Hostname":"living-5913","IPAddress":"192.168.1.33"}
18:28:14 MQT: tele/living/INFO3 = {"RestartReason":"External System"}
18:28:14 MQT: stat/living/RESULT = {"POWER1":"OFF"}
18:28:14 MQT: stat/living/POWER1 = OFF
18:28:20 CMD: syslog
18:28:20 SRC: WebConsole from 192.168.1.114
18:28:20 RSL: Topic Recibido /syslog, Tamaño de Datos 0, Datos 
18:28:20 RSL: Grupo 0, Índice 1, Comando SYSLOG, Datos 
18:28:20 MQT: stat/living/RESULT = {"SysLog":"4 (Active 4)"}
```

And all that output also was in the file` /var/log/user.log` in my RPI

@reloxx13 

So, I don't know what is the difference between my setup and yours.

What else we can test ?

ascillato, 2018-09-10:
Status 0 Output:

```
18:39:50 MQT: stat/living/STATUS = {"Status":{"Module":18,"FriendlyName":["Living"],"Topic":"living","ButtonTopic":"living","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:39:50 MQT: stat/living/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:00:18","StartupUTC":"2018-09-10T21:39:32","Sleep":0,"BootCount":33,"SaveCount":77,"SaveAddress":"3F5000"}}
18:39:50 MQT: stat/living/STATUS2 = {"StatusFWR":{"Version":"6.2.1.3","BuildDateTime":"2018-09-10T17:30:48","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
18:39:50 MQT: stat/living/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"192.168.1.104","LogPort":514,"SSId":["NetWireless","NetWireless"],"TelePeriod":300,"SetOption":["04008009","55818000","00000000"]}}
18:39:50 MQT: stat/living/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":2580,"Heap":19,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3,"Features":["00002C0A","1F80A984","0C0003B0","23021796","000000C0"]}}
18:39:50 MQT: stat/living/STATUS5 = {"StatusNET":{"Hostname":"living-5913","IPAddress":"192.168.1.33","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:59:17:19","Webserver":2,"WifiConfig":5}}
18:39:50 MQT: stat/living/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_591719","MqttUser":"homeassistant","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:39:50 MQT: stat/living/STATUS7 = {"StatusTIM":{"UTC":"Mon Sep 10 21:39:50 2018","Local":"Mon Sep 10 18:39:50 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-3,"Sunrise":"07:40","Sunset":"19:17"}}
18:39:50 MQT: stat/living/STATUS10 = {"StatusSNS":{"Time":"2018-09-10T18:39:50"}}
18:39:50 MQT: stat/living/STATUS11 = {"StatusSTS":{"Time":"2018-09-10T18:39:50","Uptime":"0T00:00:18","Vcc":2.997,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","BSSId":"18:D6:C7:80:38:2C","Channel":1,"RSSI":64}}}

```

curzon01, 2018-09-11:
I'm also unable to reproduce the issue with all the given steps above so
@reloxx13 as your project desc assumes you are using an extenend Sonoff with 4MB and own compilation - suggestion: can you try to reproduce the issue with one of the repository bin 6.1.1/6.2.1 on an original sonoff?

ascillato2, 2018-09-11:
Tried also with MQTT = 0 and also works fine:

```
00:00:00 CFG: Cargado desde flash en 3FB, Conteo 95
00:00:00 APP: Conteo Reinicios 46
00:00:00 SRC: Restart
00:00:00 Proyecto sonoff Living (Topic living, Fallback DVES_591719, GroupTopic sonoffs) Versión 6.2.1.3-2_4_2
00:00:00 CFG: Grabado a la flash en 3FA, Conteo 96, Bytes 3584
00:00:00 WIF: Probando conexión...
00:00:00 WIF: Intentando conectar...
00:00:00 WIF: Connectando a AP1 NetWireless en modo 11N como living-5913...
00:00:01 WIF: Probando conexión...
00:00:01 WIF: Intentando conectar...
00:00:02 WIF: Probando conexión...
00:00:02 WIF: Intentando conectar...
00:00:03 WIF: Probando conexión...
00:00:03 WIF: Intentando conectar...
00:00:04 WIF: Probando conexión...
00:00:04 WIF: Conectado
00:00:04 HTP: Servidor web activo en living-5913 con dirección IP 192.168.1.33
00:00:04 RSL: INFO1 = {"Module":"Generic","Version":"6.2.1.3","FallbackTopic":"DVES_591719","GroupTopic":"sonoffs"}
00:00:04 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"living-5913","IPAddress":"192.168.1.33"}
00:00:04 RSL: INFO3 = {"RestartReason":"Software/System restart"}
00:00:04 RSL: RESULT = {"POWER1":"OFF"}
00:00:04 RSL: POWER1 = OFF
00:00:07 APP: (UTC) Tue Sep 11 17:55:44 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
14:55:50 RSL: STATE = {"Time":"2018-09-11T14:55:50","Uptime":"0T00:00:13","Vcc":2.991,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","BSSId":"18:D6:C7:80:38:2C","Channel":1,"RSSI":68}}
14:55:58 HTP: Menú Principal
14:55:59 HTP: Consola
14:56:01 WIF: Probando conexión...
14:56:01 WIF: Conectado
14:56:02 CMD: syslog
14:56:02 SRC: WebConsole from 192.168.1.114
14:56:02 RSL: Topic Recibido /syslog, Tamaño de Datos 0, Datos 
14:56:02 RSL: Grupo 0, Índice 1, Comando SYSLOG, Datos 
14:56:02 RSL: RESULT = {"SysLog":"4 (Active 4)"}
```

ascillato, 2018-09-12:
@cyablo 

Please, can you test this issue again but with last firmware?
If stills happening, please share your new status 0, console output (with weblog set to 4) and also any further details of your setup as actual syslog server software, etc. Thanks

cyablo, 2018-09-12:
Im using standard rsyslogd and mosquitto on debian stretch, no special setup. No custom build tasmota...

Same with 6.2.1, after rbeoot of the tasmota device, syslog stops working until I hit "save" once under logging options.

status 0 before reboot:

> 22:34:46 CMD: status 0
22:34:46 SRC: WebConsole from 192.168.1.19
22:34:46 RSL: empfangenes topic /status, Datengröße 1, Daten 0
22:34:46 RSL: Gruppe 0, Index 1, Befehl STATUS, Daten 0
22:34:46 MQT: stat/tasmota11/STATUS = {"Status":{"Module":18,"FriendlyName":["neonroehre"],"Topic":"tasmota11","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
22:34:46 MQT: stat/tasmota11/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.26:80/sonweb/data/firmwares/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:14","StartupUTC":"2018-09-12T20:31:32","Sleep":50,"BootCount":96,"SaveCount":241,"SaveAddress":"FB000"}}
22:34:46 MQT: stat/tasmota11/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018.09.05 10:44:57","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:34:46 MQT: stat/tasmota11/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":4,"LogHost":"192.168.1.26","LogPort":514,"SSId":["xxx","xxx"],"TelePeriod":300,"SetOption":["00000009","55A18000","00000001"]}}
22:34:46 MQT: stat/tasmota11/STATUS4 = {"StatusMEM":{"ProgramSize":473,"Free":528,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","0FDAE794","24000000","23B617CE","00000000"]}}
22:34:46 MQT: stat/tasmota11/STATUS5 = {"StatusNET":{"Hostname":"tasmota11","IPAddress":"192.168.1.55","Gateway":"192.168.1.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.18","Mac":"60:01:94:51:AA:EA","Webserver":2,"WifiConfig":3}}
22:34:46 MQT: stat/tasmota11/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.26","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_51AAEA","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:34:46 MQT: stat/tasmota11/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 12 20:34:46 2018","Local":"Wed Sep 12 22:34:46 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:22","Sunset":"20:10"}}
22:34:46 MQT: stat/tasmota11/STATUS10 = {"StatusSNS":{"Time":"2018-09-12T22:34:46"}}
22:34:46 MQT: stat/tasmota11/STATUS11 = {"StatusSTS":{"Time":"2018-09-12T22:34:46","Uptime":"0T00:03:14","Vcc":3.222,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxx","RSSI":100,"APMac":"xxx"}}}

after reboot:

> 22:37:52 CMD: status 0
22:37:52 SRC: WebConsole from 192.168.1.19
22:37:52 RSL: empfangenes topic /status, Datengröße 1, Daten 0
22:37:52 RSL: Gruppe 0, Index 1, Befehl STATUS, Daten 0
22:37:52 MQT: stat/tasmota11/STATUS = {"Status":{"Module":18,"FriendlyName":["neonroehre"],"Topic":"tasmota11","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
22:37:52 MQT: stat/tasmota11/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.26:80/sonweb/data/firmwares/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:46","StartupUTC":"2018-09-12T20:37:06","Sleep":50,"BootCount":97,"SaveCount":242,"SaveAddress":"FA000"}}
22:37:52 MQT: stat/tasmota11/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018.09.05 10:44:57","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:37:53 MQT: stat/tasmota11/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":4,"LogHost":"192.168.1.26","LogPort":514,"SSId":["xxx","xxx"],"TelePeriod":300,"SetOption":["00000009","55A18000","00000001"]}}
22:37:53 MQT: stat/tasmota11/STATUS4 = {"StatusMEM":{"ProgramSize":473,"Free":528,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","0FDAE794","24000000","23B617CE","00000000"]}}
22:37:53 MQT: stat/tasmota11/STATUS5 = {"StatusNET":{"Hostname":"tasmota11","IPAddress":"192.168.1.55","Gateway":"192.168.1.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.18","Mac":"60:01:94:51:AA:EA","Webserver":2,"WifiConfig":3}}
22:37:53 MQT: stat/tasmota11/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.26","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_51AAEA","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:37:53 MQT: stat/tasmota11/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 12 20:37:53 2018","Local":"Wed Sep 12 22:37:53 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:22","Sunset":"20:10"}}
22:37:53 MQT: stat/tasmota11/STATUS10 = {"StatusSNS":{"Time":"2018-09-12T22:37:53"}}
22:37:53 MQT: stat/tasmota11/STATUS11 = {"StatusSTS":{"Time":"2018-09-12T22:37:53","Uptime":"0T00:00:47","Vcc":3.222,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxx","RSSI":100,"APMac":"xxx"}}}

ascillato, 2018-09-12:
Found the issue!!!! I could reproduce it!

when you have emulation enabled, syslog goes off. See line 1599 in sonoff.ino

So, as it is an intended behaviour in the code I will ask you to test if Alexa works when you have syslog working (doing the save in the webmenu). I don't have an Alexa to test it.

cyablo, 2018-09-12:
You mean wemo emulation? Yep, I'm using this feature for alexa. Alexa always works with wemo emulation, have not encountered a single failure of my voice commands to turn on/off a tasmota device, with and without working syslog.

Would be interesting to know the reason of this intended disabling of the remote syslog function.

ascillato, 2018-09-12:
Ok, I will propose a pull request for Theo to review it.

Thanks.

ascillato2, 2018-09-13:
Hi,

Closing as the issue has been solved. Please, ask to reopen if needed. Thanks.

closed 2018-09-13 by ascillato2